indilib/indi #507: fliusb involved in kernel oops ubuntu 4.4.0-57-generic image

gschwim, 2018-01-25:
I've been running fliusb-dkms 1.3+r3745~201801222144~ubuntu16.04.1 from the indilib ppa. This issue has been occurring for at least 7 months. The problem can be reproduced by powering the camera off while a client is connected to it. Clients tested are Ekos w/ indi and TheSkyX.  The system needs to be rebooted to recover.

Log w/ relevant output from dmesg:

```

[  654.123685] fliusb_initdev[792]: FLI USB device found: 'MicroLine ML16200'
[  654.123733] fliusb_probe[827]: FLI USB device attached; rdepaddr: 0x81; wrepaddr: 0x01; buffersize: 4096; timeout: 5000
[  772.095469] usb 5-2.3.2: USB disconnect, device number 11
[  772.095626] fliusb_disconnect[852]: FLI USB device disconnected
[  772.463179] BUG: unable to handle kernel NULL pointer dereference at 0000000000000004
[  772.463203] IP: [<ffffffff81621697>] usb_reset_configuration+0x87/0x2d0
[  772.463221] PGD 0 
[  772.463228] Oops: 0000 [#1] SMP 
[  772.463236] Modules linked in: unix_diag tcp_diag inet_diag ebtable_filter ebtables bridge drbg ansi_cprng ctr ccm arc4 ath9k_htc ath9k_common ath9k_hw ath mac80211 ftdi_sio fliusb(OE) usbserial cfg80211 cdc_acm binfmt_misc ip6table_filter ip6_tables iptable_filter ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack iptable_mangle ip_tables x_tables nls_iso8859_1 eeepc_wmi asus_wmi sparse_keymap snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic intel_rapl x86_pkg_temp_thermal intel_powerclamp snd_hda_intel crct10dif_pclmul crc32_pclmul ghash_clmulni_intel snd_hda_codec aesni_intel snd_hda_core snd_hwdep aes_x86_64 snd_pcm lrw gf128mul glue_helper ablk_helper snd_seq_midi cryptd snd_seq_midi_event snd_rawmidi input_leds serio_raw
[  772.463437]  snd_seq snd_seq_device snd_timer snd soundcore shpchp mei_me lpc_ich mei 8250_fintek tpm_infineon acpi_pad mac_hid kvm_intel kvm irqbypass nct6775 hwmon_vid coretemp 8021q garp mrp stp llc parport_pc ppdev lp parport autofs4 btrfs xor raid6_pq dm_mirror dm_region_hash dm_log hid_generic usbhid hid mxm_wmi i915 i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops e1000e ahci drm libahci ptp pps_core wmi video fjes
[  772.463557] CPU: 6 PID: 8277 Comm: indi_fli_ccd Tainted: G           OE   4.4.0-57-generic #78-Ubuntu
[  772.463576] Hardware name: ASUS All Series/Z97-A-USB31, BIOS 2801 11/11/2015
[  772.463591] task: ffff8807b181aa00 ti: ffff88079fe5c000 task.ti: ffff88079fe5c000
[  772.463606] RIP: 0010:[<ffffffff81621697>]  [<ffffffff81621697>] usb_reset_configuration+0x87/0x2d0
[  772.463626] RSP: 0018:ffff88079fe5fd88  EFLAGS: 00010246
[  772.463637] RAX: 0000000000000000 RBX: 0000000000000000 RCX: 0000000000000002
[  772.463652] RDX: 0000000080000000 RSI: 0000000000000080 RDI: ffff880817440000
[  772.463667] RBP: ffff88079fe5fdd8 R08: ffff88079fe5fdf4 R09: 000000000000ea60
[  772.463682] R10: 000000000000000f R11: 0000000000000246 R12: ffff880817440000
[  772.463696] R13: 0000000000000000 R14: 00007ffc72b452a0 R15: 000000000000ea60
[  772.463711] FS:  00007f9ac2cf5740(0000) GS:ffff88083fb80000(0000) knlGS:0000000000000000
[  772.463727] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  772.463739] CR2: 0000000000000004 CR3: 000000079fc7d000 CR4: 00000000001406e0
[  772.463754] Stack:
[  772.463758]  ffff88079fe5fdb8 ffff88079fe5fdc0 ffff88079fe5fdc8 0000000000000286
[  772.463777]  ffff880814b6a000 0000000000000002 ffff880817441000 ffff880817441018
[  772.463795]  00007ffc72b452a0 000000000000ea60 ffff88079fe5fe28 ffffffffc05e62b8
[  772.463812] Call Trace:
[  772.463821]  [<ffffffffc05e62b8>] fliusb_bulk_write+0xc8/0xd0 [fliusb]
[  772.463835]  [<ffffffffc05e7244>] fliusb_ioctl+0x454/0x5db [fliusb]
[  772.463850]  [<ffffffff8122224f>] do_vfs_ioctl+0x29f/0x490
[  772.463864]  [<ffffffff8126359b>] ? locks_lock_inode_wait+0x5b/0x1a0
[  772.463877]  [<ffffffff812224b9>] SyS_ioctl+0x79/0x90
[  772.463889]  [<ffffffff81264ca7>] ? SyS_flock+0x107/0x170
[  772.463902]  [<ffffffff818374f2>] entry_SYSCALL_64_fastpath+0x16/0x71
[  772.463916] Code: 45 d0 4d 8b ac 24 98 03 00 00 48 8b b8 28 02 00 00 e8 ce 3c 21 00 4c 89 e7 e8 06 24 ff ff 85 c0 89 45 cc 0f 85 06 02 00 00 31 db <41> 80 7d 04 00 4d 8d b5 98 00 00 00 75 36 e9 a5 01 00 00 49 8b 
[  772.463998] RIP  [<ffffffff81621697>] usb_reset_configuration+0x87/0x2d0
[  772.464013]  RSP <ffff88079fe5fd88>
[  772.464020] CR2: 0000000000000004
[  772.467142] ---[ end trace 8b217362de0110d4 ]---
```

knro, 2018-01-25:
We are waiting on FLI to release the libusb-based SDK which would render the kernel driver obsolete.

gschwim, 2018-02-03:
Appears to be fixed with new fli module per https://github.com/indilib/indi/issues/510

closed 2018-02-03 by gschwim