indilib/indi #509: OSX indi_gphoto_ccd error loading library

gschwim, 2018-02-02:
This is with whatever is shipping with Kstars 2.9.2 for OSX (Build: 2018-01-28T07:25:07Z). I cannot connect to my Sony A7S. Last I tried it was on 2.8.9 and it worked fine. I tested connection to the camera with another piece of software and had no issue. Tried a reboot as well but the problem persisted.

Debug log is as follows:

DEBUG	95.986867 sec	: Mirror lock value: 0.000000
DEBUG	95.986955 sec	: libgphoto2 info:
DEBUG	95.986978 sec	: 2.5.16
DEBUG	95.986992 sec	: all camlibs
DEBUG	95.987005 sec	: clang
DEBUG	95.987025 sec	: ltdl
DEBUG	95.987057 sec	: no EXIF
DEBUG	95.987087 sec	: Camera init. Takes about 10 seconds.
ERROR	96.005560 sec	: Camera open error (-4): Error loading a library
ERROR	96.005598 sec	: Can not open camera: Power OK? If camera is auto-mounted as external disk storage, please unmount it and disable auto-mount.

knro, 2018-02-02:
@rlancaste Is this related to the gphoto changes?

rlancaste, 2018-02-02:
Was this with the latest build on the website?  I cannot check until later today, but there was an issue for about 24 hours when i first posted the DMG but before the announcement was made.  There was a permissions issue with the gphoto camlibs and iolibs directories and they were not processed with otool properly.  I fixed that and uploaded a new dmg after a user reported it, so its possible that this build was downloaded before I fixed it?  Maybe try updating again?

gschwim, 2018-02-02:
Yep. that was the problem. Good call. All fixed. Thanks

closed 2018-02-02 by gschwim