indilib/indi #510: FLI CCD indi_fli_ccd issue ubuntu 4.13.0-32-generic

pedrodper, 2018-02-02:
Issue found when connecting to the FLI CCD using both the KStars device manager and the EKOS tool. The driver indi_fli_ccd crashes when taking a picture or setting the aperture time. Server log as obtained with the device manager:

2018-02-02T15:15:23: startup: /usr/bin/indiserver -v -p 7624 -m 100 -f /tmp/indififo79a6145a 
2018-02-02T15:15:23: listening to port 7624 on fd 3
FIFO: start indi_fli_ccd -n "FLI CCD"
With name: FLI CCD
FIFO: Starting driver indi_fli_ccd
2018-02-02T15:15:23: Driver indi_fli_ccd: pid=4345 rfd=4 wfd=7 efd=8
2018-02-02T15:15:23: Client 5: new arrival from 127.0.0.1:49004 - welcome!
2018-02-02T15:15:23: Driver indi_fli_ccd: snooping on Telescope Simulator.EQUATORIAL_EOD_COORD
2018-02-02T15:15:23: Driver indi_fli_ccd: snooping on Telescope Simulator.TELESCOPE_INFO
2018-02-02T15:15:23: Driver indi_fli_ccd: snooping on Telescope Simulator.GEOGRAPHIC_COORD
2018-02-02T15:15:23: Driver indi_fli_ccd: snooping on Focuser Simulator.ABS_ROTATOR_ANGLE
2018-02-02T15:15:23: Driver indi_fli_ccd: snooping on CCD Simulator.FILTER_SLOT
2018-02-02T15:15:23: Driver indi_fli_ccd: snooping on CCD Simulator.FILTER_NAME
2018-02-02T15:15:23: Driver indi_fli_ccd: snooping on SQM.SKY_QUALITY
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: FLIList() domain 0102
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: Entering FLIOpen
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: Trying to open file </dev/fliusb0> in domain 258.
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: Got device index 0
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: Domain: 0x0002
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli:   Type: 0x0100
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: USBDEVFS_SETCONFIGURATION return -1
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: USBDEVFS_SETCONFIGURATION return -1
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: linux_bulktransfer: attempting 2 bytes out
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: OUT      2: 00 01 
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: linux_bulktransfer: attempting 6 bytes in
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli:  IN      6: 01 00 00 00 00 19 
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: linux_bulktransfer: attempting 2 bytes out
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: OUT      2: 00 03 
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: linux_bulktransfer: attempting 25 bytes in
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli:  IN     25: 78 0d 0e 0a fe 0c c8 09 2b 00 2e 00 a6 31 b5 36 
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: linux_bulktransfer: attempting 2 bytes out
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: OUT      2: 00 02 
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: linux_bulktransfer: attempting 64 bytes in
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli:  IN     64: 4d 4c 30 31 38 34 31 31 33 00 00 00 00 00 00 00 
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: Device has following capabilities:
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli:    SUPPORTS_TDI
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli:    SUPPORTS_BGFLUSH
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli:    SUPPORTS_END_EXPOSURE
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli:    SUPPORTS_SOFTWARE_TRIGGER
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: DeviceID 10
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: SerialNum 0
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: HWRev 0100
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: FWRev 0124
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli:      Name: ML0184113
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli:     Array: (   0,   0),(3448,2574)
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli:   Visible: (  43,  46),(3369,2550)
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli:  Pix Size: (5.4e-06, 5.4e-06)
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli:     Temp.: T = AD x 1 + 0
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: Exiting FLIOpen
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: Entering FLIClose
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: Closing device index: 0 
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: Exiting FLIClose
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: Entering FLIOpen
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: Trying to open file </dev/fliusb0> in domain 258.
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: Got device index 0
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: Domain: 0x0002
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli:   Type: 0x0100
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: USBDEVFS_SETCONFIGURATION return -1
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: USBDEVFS_SETCONFIGURATION return -1
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: linux_bulktransfer: attempting 2 bytes out
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: OUT      2: 00 01 
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: linux_bulktransfer: attempting 6 bytes in
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli:  IN      6: 01 00 00 00 00 19 
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: linux_bulktransfer: attempting 2 bytes out
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: OUT      2: 00 03 
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: linux_bulktransfer: attempting 25 bytes in
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli:  IN     25: 78 0d 0e 0a fe 0c c8 09 2b 00 2e 00 a6 31 b5 36 
2018-02-02T15:15:24: Driver indi_fli_ccd: libfli: linux_bulktransfer: attempting 2 bytes out
[... 39 lines cut ...]
2018-02-02T15:15:26: Driver indi_fli_ccd: libfli:         Right Offset: 3326
2018-02-02T15:15:26: Driver indi_fli_ccd: libfli: linux_bulktransfer: attempting 2 bytes out
2018-02-02T15:15:26: Driver indi_fli_ccd: libfli: OUT      2: 00 06 
2018-02-02T15:15:26: Driver indi_fli_ccd: libfli: linux_bulktransfer: attempting 4 bytes in
2018-02-02T15:15:26: Driver indi_fli_ccd: libfli:  IN      4: 00 00 02 c3 
2018-02-02T15:15:26: Driver indi_fli_ccd: libfli: linux_bulktransfer: attempting 2 bytes out
2018-02-02T15:15:26: Driver indi_fli_ccd: libfli: OUT      2: 00 06 
2018-02-02T15:15:26: Driver indi_fli_ccd: libfli: linux_bulktransfer: attempting 4 bytes in
2018-02-02T15:15:26: Driver indi_fli_ccd: libfli:  IN      4: 00 00 02 c2 
2018-02-02T15:15:27: Driver indi_fli_ccd: libfli: linux_bulktransfer: attempting 2 bytes out
2018-02-02T15:15:27: Driver indi_fli_ccd: libfli: OUT      2: 00 06 
2018-02-02T15:15:27: Driver indi_fli_ccd: libfli: linux_bulktransfer: attempting 4 bytes in
2018-02-02T15:15:27: Driver indi_fli_ccd: libfli:  IN      4: 00 00 00 00 
2018-02-02T15:15:27: Driver indi_fli_ccd: libfli: linux_bulktransfer: attempting 65536 bytes in
2018-02-02T15:15:27: Driver indi_fli_ccd: stderr EOF
2018-02-02T15:15:27: Driver indi_fli_ccd: restart #1
2018-02-02T15:15:27: Driver indi_fli_ccd: pid=4350 rfd=4 wfd=8 efd=9
Child process 4345 died
2018-02-02T15:15:27: Driver indi_fli_ccd: snooping on Telescope Simulator.EQUATORIAL_EOD_COORD
2018-02-02T15:15:27: Driver indi_fli_ccd: snooping on Telescope Simulator.TELESCOPE_INFO
2018-02-02T15:15:27: Driver indi_fli_ccd: snooping on Telescope Simulator.GEOGRAPHIC_COORD
2018-02-02T15:15:27: Driver indi_fli_ccd: snooping on Focuser Simulator.ABS_ROTATOR_ANGLE
2018-02-02T15:15:27: Driver indi_fli_ccd: snooping on CCD Simulator.FILTER_SLOT
2018-02-02T15:15:27: Driver indi_fli_ccd: snooping on CCD Simulator.FILTER_NAME
2018-02-02T15:15:27: Driver indi_fli_ccd: snooping on SQM.SKY_QUALITY

pedrodper, 2018-02-02:
Any help is appreciated. Thank you in advance! :)

pedrodper, 2018-02-02:
Aaaand I just realized it is a kernel issue. Output of dmesg:

[ 2670.483223] ------------[ cut here ]------------
[ 2670.483230] kernel BUG at ./include/linux/scatterlist.h:90!
[ 2670.483238] invalid opcode: 0000 [#3] SMP PTI
[ 2670.484410] Modules linked in: fliusb(OE) coretemp kvm irqbypass arc4 iwldvm mac80211 snd_hda_codec_realtek snd_hda_codec_generic joydev snd_hda_intel snd_hda_codec snd_hda_core snd_hwdep input_leds serio_raw snd_pcm snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq iwlwifi lpc_ich cfg80211 snd_seq_device snd_timer snd shpchp soundcore dell_smo8800 mac_hid parport_pc ppdev lp parport autofs4 hid_generic usbhid hid i915 psmouse ahci libahci i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops drm e1000e sata_sil24 ptp pps_core video
[ 2670.487085] CPU: 1 PID: 4806 Comm: indi_fli_ccd Tainted: G      D    OE   4.13.0-32-generic #35~16.04.1-Ubuntu
[ 2670.487085] Hardware name: FUJITSU SIEMENS ESPRIMO Mobile M9410/M118DB, BIOS M118D0FR.112 - R081 - 1659 12/11/2009
[ 2670.487085] task: ffff9a807fc6c200 task.stack: ffffb6d0c176c000
[ 2670.487085] RIP: 0010:fliusb_sg_bulk_read+0x1d8/0x460 [fliusb]
[ 2670.487085] RSP: 0018:ffffb6d0c176fdb8 EFLAGS: 00010202
[ 2670.487085] RAX: fffffffffffffff2 RBX: fffffffffffffff1 RCX: ffff9a8133913038
[ 2670.487085] RDX: 0000000000000002 RSI: 0000000000000000 RDI: 00000000000005b8
[ 2670.487085] RBP: ffffb6d0c176fe10 R08: 00000000000000b8 R09: ffff9a8133911950
[ 2670.487085] R10: 0000000000000001 R11: ffff9a807fc6c200 R12: 0000000000010000
[ 2670.487085] R13: 00000000c0410680 R14: 00000000fffffff1 R15: ffff9a8133911800
[ 2670.487085] FS:  00007f03df775740(0000) GS:ffff9a813fd00000(0000) knlGS:0000000000000000
[ 2670.487085] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 2670.487085] CR2: 00007f03df761000 CR3: 0000000060548000 CR4: 00000000000006e0
[ 2670.487085] Call Trace:
[ 2670.487085]  fliusb_ioctl+0x480/0x520 [fliusb]
[ 2670.487085]  ? entry_SYSCALL_64_after_hwframe+0x149/0x168
[ 2670.487085]  ? entry_SYSCALL_64_after_hwframe+0x142/0x168
[ 2670.508011]  do_vfs_ioctl+0xa1/0x5f0
[ 2670.508011]  ? entry_SYSCALL_64_after_hwframe+0x118/0x168
[ 2670.508011]  ? entry_SYSCALL_64_after_hwframe+0x111/0x168
[ 2670.508011]  ? entry_SYSCALL_64_after_hwframe+0x10a/0x168
[ 2670.508011]  ? entry_SYSCALL_64_after_hwframe+0x103/0x168
[ 2670.508011]  ? entry_SYSCALL_64_after_hwframe+0xfc/0x168
[ 2670.508011]  ? entry_SYSCALL_64_after_hwframe+0xf5/0x168
[ 2670.508011]  ? entry_SYSCALL_64_after_hwframe+0xee/0x168
[ 2670.508011]  ? entry_SYSCALL_64_after_hwframe+0xe7/0x168
[ 2670.508011]  ? entry_SYSCALL_64_after_hwframe+0xe0/0x168
[ 2670.508011]  SyS_ioctl+0x79/0x90
[ 2670.508011]  ? entry_SYSCALL_64_after_hwframe+0xa1/0x168
[ 2670.508011]  entry_SYSCALL_64_fastpath+0x33/0xa3
[ 2670.508011] RIP: 0033:0x7f03de7c2f47
[ 2670.508011] RSP: 002b:00007ffdb5325818 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[ 2670.508011] RAX: ffffffffffffffda RBX: 0000000000000000 RCX: 00007f03de7c2f47
[ 2670.508011] RDX: 00007ffdb5325840 RSI: 000000004020f109 RDI: 0000000000000003
[ 2670.508011] RBP: 0000560d07437580 R08: 00007f03df753000 R09: 00007f03df775740
[ 2670.508011] R10: 0000000000000002 R11: 0000000000000246 R12: 00007f03df753000
[ 2670.529979] R13: 0000000000010000 R14: 00007ffdb5325d10 R15: 0000000000000082
[ 2670.529979] Code: 89 34 b9 c7 01 00 00 00 00 c7 41 04 00 10 00 00 74 20 48 8b 71 18 49 8b 54 3f 40 48 83 c1 20 48 83 c7 08 83 e6 03 f6 c2 03 74 ca <0f> 0b bb 01 00 00 00 48 8b 4d d0 4a 8d 14 21 f7 c2 ff 0f 00 00 
[ 2670.529979] RIP: fliusb_sg_bulk_read+0x1d8/0x460 [fliusb] RSP: ffffb6d0c176fdb8
[ 2670.535021] ---[ end trace 33823efe681b6166 ]---

gschwim, 2018-02-02:
Probably related to:
https://github.com/indilib/indi/issues/507

Did you get the Oops as well?

knro, 2018-02-02:
Yes, FLI reported that they fixed it in their updated kernel driver here: http://www.flicamera.com/downloads/sdk/fliusb-1.3.2.tgz

Can you try building that and checking if it resolves the problem for you? It looks like they're not ready yet to release the libusb based SDK.

pedrodper, 2018-02-02:
Thanks! I will see if the updated driver fixes the issue when I am back in the observatory on Monday. Thanks a lot for the help

gschwim, 2018-02-03:
The updated driver fixes the kernel Oops in the circumstances in which I was able to reproduce it reliably. More info on that here: https://github.com/indilib/indi/issues/507

knro, 2018-02-23:
INDI FLI package now uses libusb so no more kernel modules.

closed 2018-02-23 by knro