indilib/indi #511: indi-asi driver doesn't build under cross-compile environment

gavin-olson, 2018-02-03:
The indi-asi 3rd party driver links to a binary blob. The CMake scripts select the correct blob to link to based on the CMAKE_HOST_SYSTEM_PROCESSOR variable. If you're cross-compiling, though, this won't necessarily be correct (eg, building on a PC for a Raspberry Pi, it'll pick the x86 blob when you want one of the Arm ones). The build will fail at the linking step because the objects built from source won't match the binary.

I think the correct environment variable to use is CMAKE_SYSTEM_PROCESSOR. The problem may also affect other 3rd-party drivers which have a binary component.

knro, 2018-02-08:
Please submit a PR with the changes and I'll test it out.

gavin-olson, 2018-02-10:
Kind of new to this, but see #515

knro, 2018-02-11:
PR applied, closing this.

closed 2018-02-11 by knro