indilib/indi #512: paramount driver crash w/ backtrace

gschwim, 2018-02-04:
This has happened periodically to me, most recently about 6-7 times tonight before it just started working. No idea how or why it started working, or why it started crashing in the first place. I updated from the PPA tonight to try to resolve it but that didn't seem to help. Indiserver seems to have restarted it after a crash automatically and worked.

Backtrace:

```

*** Error in `': malloc(): memory corruption: 0x0000562082993b40 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f1e5c2ad7e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8213e)[0x7f1e5c2b813e]
/lib/x86_64-linux-gnu/libc.so.6(__libc_malloc+0x54)[0x7f1e5c2ba184]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(_Znwm+0x18)[0x7f1e5c8a3e78]
[0x562081d88c78]
[0x562081d96e62]
[0x562081d858f8]
[0x562081d85eb5]
/usr/lib/x86_64-linux-gnu/libindidriver.so.1(_ZN4INDI9Telescope8TimerHitEv+0x2d)[0x7f1e5d18493f]
/usr/lib/x86_64-linux-gnu/libindidriver.so.1(_Z9timerfuncPv+0x1f)[0x7f1e5d15a35f]
/usr/lib/x86_64-linux-gnu/libindidriver.so.1(+0x32631)[0x7f1e5d151631]
/usr/lib/x86_64-linux-gnu/libindidriver.so.1(eventLoop+0x19)[0x7f1e5d151729]
/usr/lib/x86_64-linux-gnu/libindidriver.so.1(main+0x114)[0x7f1e5d151383]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f1e5c256830]
[0x562081d81589]
======= Memory map: ========
562081d73000-562081d9d000 r-xp 00000000 08:02 21510197                   /usr/bin/indi_paramount_telescope (deleted)
562081f9d000-562081f9e000 r--p 0002a000 08:02 21510197                   /usr/bin/indi_paramount_telescope (deleted)
562081f9e000-562081f9f000 rw-p 0002b000 08:02 21510197                   /usr/bin/indi_paramount_telescope (deleted)
562082902000-5620829b8000 rw-p 00000000 00:00 0                          [heap]
7f1e54000000-7f1e54021000 rw-p 00000000 00:00 0 
7f1e54021000-7f1e58000000 ---p 00000000 00:00 0 
7f1e5aa51000-7f1e5aa58000 r-xp 00000000 08:02 26477268                   /lib/x86_64-linux-gnu/librt-2.23.so
7f1e5aa58000-7f1e5ac57000 ---p 00007000 08:02 26477268                   /lib/x86_64-linux-gnu/librt-2.23.so
7f1e5ac57000-7f1e5ac58000 r--p 00006000 08:02 26477268                   /lib/x86_64-linux-gnu/librt-2.23.so
7f1e5ac58000-7f1e5ac59000 rw-p 00007000 08:02 26477268                   /lib/x86_64-linux-gnu/librt-2.23.so
7f1e5ac59000-7f1e5ad61000 r-xp 00000000 08:02 26477079                   /lib/x86_64-linux-gnu/libm-2.23.so
7f1e5ad61000-7f1e5af60000 ---p 00108000 08:02 26477079                   /lib/x86_64-linux-gnu/libm-2.23.so
7f1e5af60000-7f1e5af61000 r--p 00107000 08:02 26477079                   /lib/x86_64-linux-gnu/libm-2.23.so
7f1e5af61000-7f1e5af62000 rw-p 00108000 08:02 26477079                   /lib/x86_64-linux-gnu/libm-2.23.so
7f1e5af62000-7f1e5af69000 r-xp 00000000 08:02 21506225                   /usr/lib/x86_64-linux-gnu/libogg.so.0.8.2
7f1e5af69000-7f1e5b169000 ---p 00007000 08:02 21506225                   /usr/lib/x86_64-linux-gnu/libogg.so.0.8.2
7f1e5b169000-7f1e5b16a000 r--p 00007000 08:02 21506225                   /usr/lib/x86_64-linux-gnu/libogg.so.0.8.2
7f1e5b16a000-7f1e5b16b000 rw-p 00008000 08:02 21506225                   /usr/lib/x86_64-linux-gnu/libogg.so.0.8.2
7f1e5b16b000-7f1e5b184000 r-xp 00000000 08:02 21506494                   /usr/lib/x86_64-linux-gnu/libtheoradec.so.1.1.4
7f1e5b184000-7f1e5b383000 ---p 00019000 08:02 21506494                   /usr/lib/x86_64-linux-gnu/libtheoradec.so.1.1.4
7f1e5b383000-7f1e5b384000 r--p 00018000 08:02 21506494                   /usr/lib/x86_64-linux-gnu/libtheoradec.so.1.1.4
7f1e5b384000-7f1e5b385000 rw-p 00019000 08:02 21506494                   /usr/lib/x86_64-linux-gnu/libtheoradec.so.1.1.4
7f1e5b385000-7f1e5b3c3000 r-xp 00000000 08:02 21506496                   /usr/lib/x86_64-linux-gnu/libtheoraenc.so.1.1.2
7f1e5b3c3000-7f1e5b5c2000 ---p 0003e000 08:02 21506496                   /usr/lib/x86_64-linux-gnu/libtheoraenc.so.1.1.2
7f1e5b5c2000-7f1e5b5c3000 r--p 0003d000 08:02 21506496                   /usr/lib/x86_64-linux-gnu/libtheoraenc.so.1.1.2
7f1e5b5c3000-7f1e5b5c4000 rw-p 0003e000 08:02 21506496                   /usr/lib/x86_64-linux-gnu/libtheoraenc.so.1.1.2
7f1e5b5c4000-7f1e5b61b000 r-xp 00000000 08:02 21506046                   /usr/lib/x86_64-linux-gnu/libjpeg.so.8.0.2
7f1e5b61b000-7f1e5b81b000 ---p 00057000 08:02 21506046                   /usr/lib/x86_64-linux-gnu/libjpeg.so.8.0.2
7f1e5b81b000-7f1e5b81c000 r--p 00057000 08:02 21506046                   /usr/lib/x86_64-linux-gnu/libjpeg.so.8.0.2
7f1e5b81c000-7f1e5b81d000 rw-p 00058000 08:02 21506046                   /usr/lib/x86_64-linux-gnu/libjpeg.so.8.0.2
7f1e5b81d000-7f1e5b836000 r-xp 00000000 08:02 26476573                   /lib/x86_64-linux-gnu/libz.so.1.2.8
7f1e5b836000-7f1e5ba35000 ---p 00019000 08:02 26476573                   /lib/x86_64-linux-gnu/libz.so.1.2.8
7f1e5ba35000-7f1e5ba36000 r--p 00018000 08:02 26476573                   /lib/x86_64-linux-gnu/libz.so.1.2.8
7f1e5ba36000-7f1e5ba37000 rw-p 00019000 08:02 26476573                   /lib/x86_64-linux-gnu/libz.so.1.2.8
7f1e5ba37000-7f1e5bb87000 r-xp 00000000 08:02 21507990                   /usr/lib/x86_64-linux-gnu/libcfitsio.so.2.3.37
7f1e5bb87000-7f1e5bd87000 ---p 00150000 08:02 21507990                   /usr/lib/x86_64-linux-gnu/libcfitsio.so.2.3.37
7f1e5bd87000-7f1e5bd88000 r--p 00150000 08:02 21507990                   /usr/lib/x86_64-linux-gnu/libcfitsio.so.2.3.37
7f1e5bd88000-7f1e5bd8a000 rw-p 00151000 08:02 21507990                   /usr/lib/x86_64-linux-gnu/libcfitsio.so.2.3.37
7f1e5bd8a000-7f1e5be01000 rw-p 00000000 00:00 0 
7f1e5be01000-7f1e5be19000 r-xp 00000000 08:02 26477219                   /lib/x86_64-linux-gnu/libpthread-2.23.so
7f1e5be19000-7f1e5c018000 ---p 00018000 08:02 26477219                   /lib/x86_64-linux-gnu/libpthread-2.23.so
7f1e5c018000-7f1e5c019000 r--p 00017000 08:02 26477219                   /lib/x86_64-linux-gnu/libpthread-2.23.so
7f1e5c019000-7f1e5c01a000 rw-p 00018000 08:02 26477219                   /lib/x86_64-linux-gnu/libpthread-2.23.so
7f1e5c01a000-7f1e5c01e000 rw-p 00000000 00:00 0 
7f1e5c01e000-7f1e5c035000 r-xp 00000000 08:02 26481280                   /lib/x86_64-linux-gnu/libusb-1.0.so.0.1.0
7f1e5c035000-7f1e5c234000 ---p 00017000 08:02 26481280                   /lib/x86_64-linux-gnu/libusb-1.0.so.0.1.0
[... 17 lines cut ...]
7f1e5d11d000-7f1e5d11e000 r--p 00385000 08:02 21507988                   /usr/lib/x86_64-linux-gnu/libnova-0.14.so.0.0.0
7f1e5d11e000-7f1e5d11f000 rw-p 00386000 08:02 21507988                   /usr/lib/x86_64-linux-gnu/libnova-0.14.so.0.0.0
7f1e5d11f000-7f1e5d204000 r-xp 00000000 08:02 21496677                   /usr/lib/x86_64-linux-gnu/libindidriver.so.1.6.3 (deleted)
7f1e5d204000-7f1e5d404000 ---p 000e5000 08:02 21496677                   /usr/lib/x86_64-linux-gnu/libindidriver.so.1.6.3 (deleted)
7f1e5d404000-7f1e5d407000 r--p 000e5000 08:02 21496677                   /usr/lib/x86_64-linux-gnu/libindidriver.so.1.6.3 (deleted)
7f1e5d407000-7f1e5d409000 rw-p 000e8000 08:02 21496677                   /usr/lib/x86_64-linux-gnu/libindidriver.so.1.6.3 (deleted)
7f1e5d409000-7f1e5d472000 rw-p 00000000 00:00 0 
7f1e5d472000-7f1e5d498000 r-xp 00000000 08:02 26477218                   /lib/x86_64-linux-gnu/ld-2.23.so
7f1e5d640000-7f1e5d645000 rw-p 00000000 00:00 0 
7f1e5d645000-7f1e5d663000 r-xp 00000000 08:02 26477208                   /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f1e5d663000-7f1e5d664000 r--p 0001d000 08:02 26477208                   /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f1e5d664000-7f1e5d665000 rw-p 0001e000 08:02 26477208                   /lib/x86_64-linux-gnu/libudev.so.1.6.4
7f1e5d665000-7f1e5d66b000 rw-p 00000000 00:00 0 
7f1e5d696000-7f1e5d697000 rw-p 00000000 00:00 0 
7f1e5d697000-7f1e5d698000 r--p 00025000 08:02 26477218                   /lib/x86_64-linux-gnu/ld-2.23.so
7f1e5d698000-7f1e5d699000 rw-p 00026000 08:02 26477218                   /lib/x86_64-linux-gnu/ld-2.23.so
7f1e5d699000-7f1e5d69a000 rw-p 00000000 00:00 0 
7ffff3f0f000-7ffff3f30000 rw-p 00000000 00:00 0                          [stack]
7ffff3f4f000-7ffff3f51000 r--p 00000000 00:00 0                          [vvar]
7ffff3f51000-7ffff3f53000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
2018-02-04T03:19:35: Driver indi_paramount_telescope: stderr EOF
2018-02-04T03:19:35: Driver indi_paramount_telescope: restart #1
Child process 15396 died
```

knro, 2018-02-04:
When does it crash? When you start it immediately or when you connect? Can you run it via GDB and then connect to it from KStars?

`gdb --eval-command="set follow-fork-mode child" --args indiserver -v indi_paramount_telescope indi_simulator_ccd
`
Then in KStars, connect to it directly or via Ekos. If it crashes, type **bt** to get a useful backtrace. Or if you know how to reliably reproduce the crash, let me know.

gschwim, 2018-02-04:
If I have kstars start the indiserver the crash occurs within about 10 seconds of it starting and connecting. If I start the indiserver outside of kstars the indiserver seems to run until the connect from kstars occurs. I can't reproduce it reliably. 

I'll have a go at running GDB.

gschwim, 2018-02-04:
Captured a backtrace with GDB. Output attached.

I got lucky, it happened on the 2nd try. I wasn't able to reproduce it again after another 10 tries. It seems to happen sporadically. Restarting TSX and the mount didn't help when it was happening back-to-back.
[paramount_bt.log](https://github.com/indilib/indi/files/1693139/paramount_bt.log)

I also noticed that Ekos sometimes doesn't notice the crash, and thinks things are OK. The indi manager shows Ekos is connected, but it's not getting a real state from the mount. TSX shows no connection as well. I think this has something to do with the restart of the driver. It may not always restart correctly, but I'm not sure as I can't reproduce it reliably.

gschwim, 2018-02-06:
Testing some more. I seem to be able to stop the crash once by removing the ~/.indi/Paramount_config.xml and ~/.indi/Paramount_config.xml.default. On restart I have to put the IP/port in. It connects fine. If I restart kstars and connect again the paramount driver crashes a lot of the time but not all the time.

knro, 2018-02-06:
Crash happens in regex processing of RA/DEC. Do you have Ekos mount logging turn on? what's the last thing you get from the mount there before it crashes?

gschwim, 2018-02-06:
Here's the log from connection to crash as seen from the ekos log. The last line seems to coincide with the crash. Looks unusual, like part of the indi config leaked in somehow?

[ekos.log](https://github.com/indilib/indi/files/1698059/ekos.log)

knro, 2018-02-06:
Thanks. Can you git pull and compile and test again?

gschwim, 2018-02-11:
Crash appears to be resolved. Thanks for the fix! Closing.

closed 2018-02-11 by gschwim