indilib/indi #514: indi_asi_wheel ASI EFW SelectFilter(): no fw_id

d33psky, 2018-02-08:
indi_asi_wheel from todays' PPA asi-common 0.8.0+r3461~201710091055~ubuntu16.04.1

indiserver with indi_asi_wheel on 1 system
kstars/ekos on another system

Sometimes this happens where the filter wheel cannot be controlled anymore :
2018-02-08T22:19:39: [INFO] SelectFilter(): no fw_id

I've seen this happen about 10 times now.

grep EFW ./logs/2018-02-08/log_23-18-38.txt
```
[2018-02-08T23:19:01.156 CET DEBG ][           org.kde.kstars.indi] - Received new device ASI EFW
[2018-02-08T23:19:01.173 CET DEBG ][           org.kde.kstars.indi] - INDIListener: New device ASI EFW
[2018-02-08T23:19:01.173 CET INFO ][           org.kde.kstars.ekos] - Ekos received a new device:  ASI EFW
[2018-02-08T23:19:01.180 CET DEBG ][           org.kde.kstars.indi] - < ASI EFW >: < CONNECTION >
[2018-02-08T23:19:01.189 CET DEBG ][           org.kde.kstars.indi] - < ASI EFW >: < DRIVER_INFO >
[2018-02-08T23:19:01.199 CET DEBG ][           org.kde.kstars.indi] - < ASI EFW >: < DEBUG >
[2018-02-08T23:19:01.204 CET DEBG ][           org.kde.kstars.indi] - < ASI EFW >: < SIMULATION >
[2018-02-08T23:19:01.207 CET DEBG ][           org.kde.kstars.indi] - < ASI EFW >: < CONFIG_PROCESS >
[2018-02-08T23:19:01.208 CET DEBG ][           org.kde.kstars.indi] - < ASI EFW >: < DEBUG_LEVEL >
[2018-02-08T23:19:01.209 CET DEBG ][           org.kde.kstars.indi] - < ASI EFW >: < LOGGING_LEVEL >
[2018-02-08T23:19:01.209 CET DEBG ][           org.kde.kstars.indi] - < ASI EFW >: < LOG_OUTPUT >
[2018-02-08T23:19:01.301 CET INFO ][           org.kde.kstars.ekos] - ASI EFW is connected.
[2018-02-08T23:19:01.348 CET DEBG ][           org.kde.kstars.indi] - < ASI EFW >: < FILTER_SLOT >
[2018-02-08T23:19:01.348 CET INFO ][           org.kde.kstars.ekos] - "ASI EFW filter is online."
[2018-02-08T23:19:01.518 CET DEBG ][           org.kde.kstars.indi] - < ASI EFW >: < FILTER_NAME >
[2018-02-08T23:19:01.522 CET DEBG ][           org.kde.kstars.indi] - < ASI EFW >: < USEJOYSTICK >
[2018-02-08T23:19:01.523 CET DEBG ][           org.kde.kstars.indi] - ASI EFW : "[DEBUG] Toggle Debug Level -- Driver Debug "
[2018-02-08T23:19:39.909 CET INFO ][           org.kde.kstars.indi] - ASI EFW :  "[INFO] Setting current filter to slot 2 "
[2018-02-08T23:19:39.910 CET INFO ][           org.kde.kstars.indi] - ASI EFW :  "[INFO] SelectFilter(): no fw_id "
```

INDI control panel EFW disconnect and reconnect does not help.

knro, 2018-02-09:
The package is from indi-asi, not asi-common. At any rate, I added one more debug statement to the EFW driver that could shed light on this as fw_id should not be < 0 if enumeration worked fine. Git pull and compile again or wait for tomorrow's PPA update.

d33psky, 2018-02-16:
The issue is still there, but now it's detected earlier :
 ```2018-02-16T11:58:34: [INFO] Connect(): no filter wheel known, fw_id = -1```
which at least prevents kstars from using the filter wheel during a focus framing start. [1]

INDI control panel disconnect and reconnect  does not help.
Several hours ago this same running indi_asi_wheel instance had a working connection to the wheel.
Stopping indi_server and starting it makes everything work again. (At least for a while it seems)

ps.
[1] https://bugs.kde.org/show_bug.cgi?id=390107 # crash on focus framing loop start

d33psky, 2018-02-16:
https://github.com/indilib/indi/pull/521 fixes this issue

d33psky, 2018-02-20:
fixed by #521

closed 2018-02-20 by d33psky