indilib/indi #529: Ekos complains of exposure timeout, indi_sx_ccd crashes at shutdown

gschwim, 2018-02-27:
Ekos complains about the guider exposure timing out:

`2018-02-27T00:38:49 Exposure timeout. Restarting exposure...`

The main imager continues operating during this period. On shutting down the session to restart, indi_sx_ccd crashed. 

I saw this occur after the PPA build problem documented in https://github.com/indilib/indi/issues/526 when I re-built the driver from github to get it working. The PPA update fixed the original problem but this one seems to still exist.

[guider_timeout.zip](https://github.com/indilib/indi/files/1761726/guider_timeout.zip)

knro, 2018-02-27:
The log does not contain any "Exposure Timeout"?

gschwim, 2018-02-27:
Correct. I don't see that explicit message in the log. However, I do see this which correlates to the message Ekos is giving above. The guide exposures are 10sec, yet this one goes ~34s before receiving the frame (without context for clarity):

`[2018-02-27T00:38:16.462 MST DEBG ][     org.kde.kstars.ekos.guide] - Capturing frame...
[2018-02-27T00:38:50.045 MST DEBG ][     org.kde.kstars.ekos.guide] - Received guide frame.
`
...and with context entries from indi:

`[2018-02-27T00:38:16.462 MST DEBG ][     org.kde.kstars.ekos.guide] - Capturing frame...
[2018-02-27T00:38:16.544 MST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2018-02-27T07:38:16: Driver indi
_sx_ccd: sxClearPixels: libusb_control_transfer -> OK"
[2018-02-27T00:38:50.007 MST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2018-02-27T07:38:49: Driver indi
_sx_ccd: libfli:  IN     14sxClearPixels: libusb_control_transfer -> OK"
[2018-02-27T00:38:50.007 MST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2018-02-27T07:38:49: Driver indi
_sx_ccd: sxLatchPixels: libusb_control_transfer -> OK"
[2018-02-27T00:38:50.007 MST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2018-02-27T07:38:49: Driver indi_sx_ccd: sxReadPixels: libusb_control_transfer -> OK"
[2018-02-27T00:38:50.045 MST DEBG ][     org.kde.kstars.ekos.guide] - Received guide frame.
`
Here's another one at ~30sec with context. The 2nd line is ~19sec after the 1st:

`[2018-02-27T00:36:00.151 MST DEBG ][     org.kde.kstars.ekos.guide] - Capturing frame...
[2018-02-27T00:36:19.430 MST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2018-02-27T07:36:19: Driver indi
_sx_ccd: sxClearPixels: libusb_control_transfer -> OK"
[2018-02-27T00:36:20.658 MST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2018-02-27T07:36:20: Driver indi
_sx_ccd: sxClearPixels: libusb_control_transfer -> OK"
[2018-02-27T00:36:27.501 MST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2018-02-27T07:36:27: Driver indi
_sx_ccd: sxClearPixels: libusb_control_transfer -> OK"
[2018-02-27T00:36:30.491 MST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2018-02-27T07:36:30: Driver indi
_sx_ccd: sxLatchPixels: libusb_control_transfer -> OK"
[2018-02-27T00:36:30.758 MST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2018-02-27T07:36:30: Driver indi
_sx_ccd: libfli: libusb_bulktransfer: attempting 14 bytes insxReadPixels: libusb_control_transfer -> OK"
[2018-02-27T00:36:30.779 MST DEBG ][     org.kde.kstars.ekos.guide] - Received guide frame.
`

gschwim, 2018-02-27:
Digging on this a bit more. In the indi_sx_ccd driver I had enabled image upload to "Both" such that I'd get a copy of the images on disk as well as them being presented to ekos. Looking at one of the cases from the logs above, I see this:

```
T00:38:16.462 MST DEBG ][     org.kde.kstars.ekos.guide] - Capturing frame...
T00:38:26 - FIT arrives on disk, header confirms image started at T00:38:16.501.
T00:38:49 Exposure timeout. Restarting exposure... (Ekos console)
T00:38:50.045 MST DEBG ][     org.kde.kstars.ekos.guide] - Received guide frame.
```
According to this the image arrived on time. Perhaps the indi driver is doing the "right thing" and Ekos is having a problem? Of note - I have debugging enabled to be able to log a lot of this. The log files are absolutely huge. Maybe the size of the log file is causing problems given that odd behaviors start up after a long run time. </speculation>

knro, 2018-03-03:
Are you still getting this? The recent changes I've made to KStars would make the extensive logging not a big problem since it's offloaded to a temporary file.

gschwim, 2018-03-03:
Yeah, I got it one more time last night and indi_sx_ccd crashed right after I shut down. Logging was at a minimum. I need to keep testing.  Why not close it or assign it to me while I continue investigating?

k-ross, 2018-05-24:
I also get (at least part of) this problem.  I run INDI on a Raspberry Pi over WiFi.  If I tell the image capture to upload files back to Ekos, then the guider times out and aborts whenever the capture module is done with an image and uploads back to Ekos.  I get around that problem by telling INDI to save locally only (on the Pi).  Of course doing that exposed another bug, which I submitted a fix for. :)

knro, 2018-10-22:
I cannot reproduce this with git code. Can you please check?