indilib/indi #532: asi178(main) asi120(guide) -guide fires main in focus page.

Blueshawk, 2018-02-28:
The two cameras appear to be interacting somehow. When I guide internally or even with PHD2 the asi120 image pops up on the preview when trying to focus the asi178 despite all efforts to stop it. 
When guiding The 120 will actually cause the 178 to capture in the focus tab showing the buttons as pushed.  The asi178 also will not capture in the photo tab while this is going on.  
To reproduce, attach two asi cams and try to guide with one.
Related? Disconnecting either camera now results in kstars/ekos/indi control to lock up with a single cpu 100% zombie race. Nothing in journal or dmesg. Killall kstars and restarting everything is needed. 

I appreciate all the work you guys do on this stuff. After all we're all in it for the hugs. :1st_place_medal:

knro, 2018-02-28:
I have the exact same cameras (178 && 120 on StellarMate) and I just tried them with internal guider + PHD2 and didn't see any issues or interaction between the two. So it is most likely not an INDI driver issue. Are you using KStars 2.9.3?

kecsap, 2018-03-01:
I recently started to use two ASI cameras and PHD2 external guiding. I had some issues, the most important things to check:
- Use the latest INDI/KStars
- Select the ZWO driver for both CCD and Guider device in INDI settings
- Before connecting PHD2, be sure that you chosen the correct camera from the ASI driver for guiding.
- In the INDI device settings, check if the Telescope Primary vs. Guide setting is selected for both ASI cameras correctly on the Options tab.
- Be sure that the correct camera is selected for guiding on the Guide module.
- Be sure "Receive external guide frames" option is NOT selected in Guide module/Options/Guide.

Blueshawk, 2018-03-01:
Hi @knro  You've been getting so busy lately. :D Thanks for checking it out. 
@kecsap Thanks, I'll check that last option, the rest checks out. 

I've got Build: 2018-02-02T08:48:40Z which says 2.9.2 here on my notebook.  I'll take a look and track down the update for my main pc later when I get home and retest it to see if I can narrow down what triggered it. Most of my system is new or updated at least as of last week(see below). 

  It could be procedural. In at least one case I left framing on in the focus module and started guiding before things got weird. In an earlier session this happened when I was trying to use PHD2 and I thought it was because I left the camera connected and running in the focus module when I started the guide loop.
Unfortunately I can't retest till this rain stops due to the need for a guide star. 

Also probably not related but worth reporting, I haven't been able to compile from git for a few days due to a library dependency issue (something 1.2.8) and have been using a 2 week+ old build as a result. I'm going to start throttling back on that anyway so I can get the system stable, as discussed branch/ppa  discussion in issue #435  but thought I'd better mention it. I'll set an issue for it when I can get the particulars and reproduce it - might wait till after the new odroid-xu4 is set up as things are bound to build different.

thanks!
Ray

Blueshawk, 2018-07-17:
Following up on this, I tried something I read that said when using two zwo cameras to only put the main camera in the hardware setup page and leave the guide cam slot blank. That seems to be working better, but I haven't had long enough sessions to try phd2 to test for the problem there. Other testing(indi-onstep) has taken precedence in the short runs I've gotten this summer. I'll update this(or make a new one) is if I see any more weirdness and try to nail down what was doing it.

closed 2018-07-17 by Blueshawk