indilib/indi #535: fli readout differences between indi & non-indi clients, power cycle differences

gschwim, 2018-03-03:
When creating a master bias I noticed a big difference between frames captured via indiserver (both ekos and PixInsight clients) and TheSkyX. Testing this I focused on bias frames, however I am seeing this in the other frame types as well. 

The frames from TSX look clean, whereas the indiserver captured frames have banding. However frames captured via indiserver _after_ TSX has been allowed to connect to the camera look clean as long as the camera is not power cycled. If I power cycle the camera the banding returns. Guessing at this it would seem that TSX may be making some adjustment to the camera that indiserver can take advantage of until the camera is power cycled. What's interesting is TSX is using the same fliusb driver from the PPA.

Statistics on the pre and post TSX connection frames differ as well, with the post TSX frames resembling the TSX captured frames more closely. 

The camera is an FLI ML16200. I'm using fliusb from the PPA for TSX and indi_fli from the fli_work branch of the git repository. The master bias is the exception as noted.

Sample files linked to below. Inventoried as:

- ml16200_bias_ek_1x1_-25c_02.2018 - master bias captured using fliusb (pre fli_work branch testing)
- ekos-freshstart* - ekos bias frames after camera power cycle
- PI_freshstart* - PixInsight bias frame captured after camera power cycle
- TSX.* - TSX bias frames
- ekos-afterTSX* - ekos bias frames captured after TSX was used, camera _not_ power cycled
- PI_afterTSX* - PixInsight bias frames captured after TSX was used, camera _not_ power cycled

[Example Files Download](https://www.dropbox.com/sh/57ocdxjl2wf0tdl/AAAgBNfubzvD52rnD6iLLJPna?dl=0)

knro, 2018-03-03:
fliusb is no longer used in INDI since the move to libusb. Can you check if this problem only surfaced when we switched to using to libusb-based FLI driver?

git checkout 48f82237b5e276184ec9d1a77c64e09a5081739e

Then compile & install libfli followed by compile & install of indi-fli.

gschwim, 2018-03-03:
Thx for the quick reply! 

I may have libfli and fliusb mixed up. I was running the fliusb update per https://github.com/indilib/indi/issues/510. indiserver was using indi-fli from the fli_work branch we were testing last week or so.

In any case I checked out 48f8223 and installed as suggested. The banding is still there. I added some samples as ekos-updated-indi* in the download folder provided above.

knro, 2018-03-03:
Ok it is probably due to the fact that INDI FLI driver sets minimum duration to 50ms. Can you edit the driver and change this line:

if ((err = FLISetExposureTime(fli_dev, (long)(duration * 1000))))

to this

if ((err = FLISetExposureTime(fli_dev, 0)))

I'm curious to see if passing 0 would go through or not.

gschwim, 2018-03-03:
OK. This is the result I get:

`2018-03-03T17:00:34: [ERROR] Requested exposure value (0) seconds out of bounds [0.01,3600]. `
This appears to be set in indibase/indiccd.cpp. 

Before we adjust that though I can see how a minimum exposure _may_ have an impact on bias but not on other frame types. I'm seeing it across the board - bias, dark, flat, light. Should rule out minimum exposure times as a factor, no? Also, connecting to the camera with TSX then switching back to indi wouldn't "fix" the problem if it was related to minimum exposure times I think. I have to wonder what TSX might be doing differently when it sets the camera up on connection.

knro, 2018-03-03:
I made changes to INDI FLI on **master** to introduce controls for flushing. I believe those are the reason for this behavior. Try to enable background flushing first under options and see if that works, it does not work on all models. If it works, then test the image again. If it doesn't work, try increasing the number of N pre-exposure flushes. If that improves the situation, let me know what value was appropriate so I use it as default.

gschwim, 2018-03-03:
OK. Switched to master and tried to build libfli and indi-fli.

```
git checkout master
git reset --hard
```

libfli builds with warnings but installs:

```
schwim@observatory ~/src/build/indi_3rdparty/libfli $ make
-- Configuring done
-- Generating done
-- Build files have been written to: /home/schwim/src/build/indi_3rdparty/libfli
Scanning dependencies of target fli
[  7%] Building C object CMakeFiles/fli.dir/libfli.c.o
[ 14%] Building C object CMakeFiles/fli.dir/libfli-camera.c.o
/home/schwim/src/indi/3rdparty/libfli/libfli-camera.c: In function ‘fli_camera_command’:
/home/schwim/src/indi/3rdparty/libfli/libfli-camera.c:990:19: warning: variable ‘cam’ set but not used [-Wunused-but-set-variable]
     flicamdata_t *cam;
                   ^
[ 21%] Building C object CMakeFiles/fli.dir/libfli-camera-parport.c.o
[ 28%] Building C object CMakeFiles/fli.dir/libfli-camera-usb.c.o
/home/schwim/src/indi/3rdparty/libfli/libfli-camera-usb.c: In function ‘fli_camera_usb_read_eeprom’:
/home/schwim/src/indi/3rdparty/libfli/libfli-camera-usb.c:2944:7: warning: variable ‘r’ set but not used [-Wunused-but-set-variable]
  long r = 0;
       ^
/home/schwim/src/indi/3rdparty/libfli/libfli-camera-usb.c: In function ‘fli_camera_usb_write_eeprom’:
/home/schwim/src/indi/3rdparty/libfli/libfli-camera-usb.c:3000:7: warning: variable ‘r’ set but not used [-Wunused-but-set-variable]
  long r = 0;
       ^
[ 35%] Building C object CMakeFiles/fli.dir/libfli-filter-focuser.c.o
/home/schwim/src/indi/3rdparty/libfli/libfli-filter-focuser.c: In function ‘fli_homedevice’:
/home/schwim/src/indi/3rdparty/libfli/libfli-filter-focuser.c:1252:11: warning: variable ‘begin’ set but not used [-Wunused-but-set-variable]
   clock_t begin;
           ^
[ 42%] Building C object CMakeFiles/fli.dir/libfli-mem.c.o
[ 50%] Building C object CMakeFiles/fli.dir/libfli-raw.c.o
[ 57%] Building C object CMakeFiles/fli.dir/unix/libfli-usb.c.o
[ 64%] Building C object CMakeFiles/fli.dir/unix/libfli-debug.c.o
[ 71%] Building C object CMakeFiles/fli.dir/unix/libfli-serial.c.o
[ 78%] Building C object CMakeFiles/fli.dir/unix/libfli-sys.c.o
[ 85%] Building C object CMakeFiles/fli.dir/unix/libusb/libfli-usb-sys.c.o
/home/schwim/src/indi/3rdparty/libfli/unix/libusb/libfli-usb-sys.c: In function ‘libusb_fli_create_name’:
/home/schwim/src/indi/3rdparty/libfli/unix/libusb/libfli-usb-sys.c:384:11: warning: variable ‘addr’ set but not used [-Wunused-but-set-variable]
   uint8_t addr, port_nums[7];
           ^
[ 92%] Building C object CMakeFiles/fli.dir/unix/linux/libfli-parport.c.o
[100%] Linking C shared library libfli.so
[100%] Built target fli
schwim@observatory ~/src/build/indi_3rdparty/libfli $ sudo make install
[sudo] password for schwim:
[100%] Built target fli
Install the project...
-- Install configuration: "Debug"
-- Installing: /usr/include/libfli.h
-- Installing: /usr/lib/x86_64-linux-gnu/libfli.so.2.0
-- Up-to-date: /usr/lib/x86_64-linux-gnu/libfli.so.2
-- Installing: /usr/lib/x86_64-linux-gnu/libfli.so
-- Installing: /lib/udev/rules.d/99-fli.rules
```

indi-fli fails to build:

```
schwim@observatory ~/src/build/indi_3rdparty/indi-fli $ make
-- Found CFITSIO: /usr/lib/x86_64-linux-gnu/libcfitsio.so
-- Found libfli: /usr/lib/x86_64-linux-gnu/libfli.so
-- Configuring done
-- Generating done
-- Build files have been written to: /home/schwim/src/build/indi_3rdparty/indi-fli
Scanning dependencies of target indi_fli_wheel
[ 16%] Building CXX object CMakeFiles/indi_fli_wheel.dir/fli_cfw.cpp.o
/home/schwim/src/indi/3rdparty/indi-fli/fli_cfw.cpp: In member function ‘virtual bool FLICFW::updateProperties()’:
/home/schwim/src/indi/3rdparty/indi-fli/fli_cfw.cpp:129:28: error: ‘POLLMS’ was not declared in this scope
         timerID = SetTimer(POLLMS);
                            ^
CMakeFiles/indi_fli_wheel.dir/build.make:62: recipe for target 'CMakeFiles/indi_fli_wheel.dir/fli_cfw.cpp.o' failed
make[2]: *** [CMakeFiles/indi_fli_wheel.dir/fli_cfw.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/indi_fli_wheel.dir/all' failed
make[1]: *** [CMakeFiles/indi_fli_wheel.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

knro, 2018-03-03:
need to compile and install libindi first

gschwim, 2018-03-07:
Thought I replied but apparently I not.

I build everything as requested. I see no setting for background flushing. Ran a few tests and things appear to be unchanged.

knro, 2018-03-07:
Please pull now and try again.

gschwim, 2018-03-07:
OK, I see the settings there now. I cannot change the N Flush value. Doesn't seem to be a setting box, just a value.

At first glance even without enabling BKG Flush there seems to be a difference. I'll continue to test. 

I have gotten a strange frame now and again with the update. I haven't figured out how to reproduce it yet though. Is BKG Flush enabling RBI Flooding? I've never used it before (even with TSX - never needed it). 

https://www.dropbox.com/s/0fecry30huyv1sg/noflush_brightspot.fits?dl=0

I also reached out to FLI to see if they can provide any insight into what TSX might be doing differently.

knro, 2018-03-08:
Opps sorry for some reason that property was set to RO, it's now RW so please try again. The reason is there seems to be a difference is because I set the exposure time to 0 when taking flats. I wasn't sure if this was going to be accepted by the SDK but apparently it does.

gschwim, 2018-03-08:
OK, that got it going. Will do some testing over the next few days.

knro, 2018-03-16:
How did the testing go? any other issues?

gschwim, 2018-03-17:
Initial frames look ok but I need to test more. Been traveling, hope to get back to it next week.

gschwim, 2018-03-29:
Sorry for the delay. I ran a couple of tests tonight. First thing I noticed is at defaults (background flush disabled) I get a bright spot at the upper left corner of the frame, no matter the exposure. If I enable background flush it goes away (N flush untouched at 0). 

Example images, one with background flush enabled and one without, plus a 60s dark with background flush enabled.

https://www.dropbox.com/sh/6g2yyjwja7h3nm7/AAD7KPbY_32U7E2TVjOGNkawa?dl=0

Is something going on with the RBI flood? Never had to use it with this camera before. Not sure how it works.

I still need to test for the banding further. However, I do see a hint of it in these images.

knro, 2018-03-29:
So it appears this option needs to be on by default, if supported by the camera. Let me know the results of any further tests. Thanks!

gschwim, 2018-03-30:
Hmm. Interesting that the flash in the upper left corner did not occur prior to the additions of the background flush capability. Any ideas what might have caused that? I noticed that if I enable background flush, it goes away. If I then kill the indi server and restart it (without having saved the config) it is turned off again and the bright spot is back. Maybe it's on by default and the software is shutting it off?

In any case, I tested with NFlushes from 0-16 and the banding is still there. Also relevant is the bands seem to move.

knro, 2018-04-10:
Should background flushing be enabled by default then?

Regarding BIAS frame, what's the reported exposure time in TSX? Could it be that what you're seeing is autostretch artifact.. have you compared the two fits in another program like Pixinsight?

gschwim, 2018-04-12:
It could be that background flush needs to be on by default. I'm talking with FLI and will provide any info they have.

The BIAS frame exposure time in TSX is listed as '0'. 

I don't think this is a stretch artifact. I'm looking at all of these in PixInsight. The median background values differ so there is a difference somewhere.

I just tried a new test where I used TSX to control the download mode of the camera (12mhz vs. 2mhz) then captured with indi/ekos. This may be a contributing factor as indi/ekos seems to follow whatever mode is set by TSX. 

I don't see any means of setting the download mode/speed for FLI in indi/ekos. Maybe this just needs to be added?

gschwim, 2018-04-12:
Yeah, looked through that too, didn't see it. Notably, the date in that doc is from 2005 so its a bit, uh, old.

I have an email going with FLI. Will copy you in for the discussion.

gschwim, 2018-04-13:
Regarding the download speed of the camera, I received this email from Greg Terrance @ FLI:

> The functions for speed changing are:
> 
>      LIBFLIAPI FLIGetCameraModeString(flidev_t dev, flimode_t mode_index, char *mode_string, size_t siz)
> 
> This function gets a string describing a mode offered by the camera. You would call this function repeatedly with incrementing indexes to get a list of the modes the camera supports (some cameras have several modes).  This function fails when mode_index is too large (you have selected all the possible modes).
> 
>      LIBFLIAPI FLIGetCameraMode(flidev_t dev, flimode_t *mode_index)
> 
> This function gets the index of the mode the camera is currently in.
> 
>      LIBFLIAPI FLISetCameraMode(flidev_t dev, flimode_t mode_index)
> 
> This function sets the camera mode.

Hopefully this solves the problem completely. I can provide access to the camera for you to test if that would be helpful.

knro, 2018-04-13:
I just pushed some code, can you please rebuild and see if it works?

gschwim, 2018-04-13:
Rebuilt and initially tested. The speed mode now changes properly. Re: the original banding problem I am going to need to do some more testing to confirm final resolution.

Adding in info from Jim Moronski @ FLI with my own comments:

 - Background Flush should be left as is at camera power-on for cameras newer than 2005.
 - Mode changes require either: A) capture a "throw away" bias frame after the change, or B) waiting at least as long as a frame download in the new mode before capturing an image. Since each FLI camera may behave differently it might be best to use option 'A' so as to ensure the switch is complete before any other imaging operation is completed. Can we do this in indi_fli_ccd?

knro, 2018-04-14:
I added a warning asking the user to take a bias frame to sync the change, and BG flush is off by default.

gschwim, 2018-04-15:
OK, I see that change. Waiting the download time of the new mode seems to do the trick. Maybe just put a 20 second timer in before the mode is shown to be complete? (e.g. count 20s before turning the light green in the control panel, or lock access to the camera for 20 seconds while the timer counts down).

Still testing for banding but results look promising.

gschwim, 2018-06-27:
Update - The download setting mode seems to do the trick. One problem is the setting does not persist in the configuration. For example, after power cycling the camera and restarting the indi server, the camera reverts back to focus mode. The user needs to remember to go adjust the camera mode every time this happens (daily) else you end up with bad data again.

commit a4539340 (2018-06-28): Remember download mode.

Should fix #535
closed 2018-06-28 by knro