indilib/indi #539: xagyl filter wheel fails to load XML

Blueshawk, 2018-03-12:
I've been struggling with a problem on my rpi3 server for a while now but waited till I finished upgrades in hopes ot would go away on its own..it hasn't. While I can run the filter wheel just fine on my backup server,(my work notebook) it just won't load the filter_id and slots from xml and truncates it if I save. 
 On the older rpi3 I had tried to bisect it to see where the problem was introduced but it appears to not be in indi_xagyl_wheel which hasn't changed much.
Then I compiled from git on a brand new system running armbian on an Odroid-xu4 and it did exactly the same thing. 
Kstars is updated on both local and remote, indilib full build from git master on odroid. It does this both on remote and local connections to this unit, as well as the rpi3b, but not with the backup notebook computer.  There's also a typo at "au[t]o search off" 
 It also turns the debug on then off again at about the time the failure happens, but I think that's normal?
Anybody seen something similar to this corruption of xml loading?
Thanks, 
Ray

Blueshawk, 2018-03-12:
This happens on connect. Pretty sure this is why things are failing but I haven't been able to track down why it's happening. 

2018-03-12T14:34:48: Client 0: new arrival from 10.0.0.32:54864 - welcome!
*** stack smashing detected ***:  terminated
2018-03-12T14:34:53: Driver indi_xagyl_wheel: stderr EOF
Child process 5446 died
2018-03-12T14:34:53: Driver indi_xagyl_wheel: restart #1
2018-03-12T14:34:53: Driver indi_xagyl_wheel: pid=5451 rfd=5 wfd=14 efd=17

knro, 2018-03-12:
that's a crash alright. Is this from the GIT version or PPA?

Blueshawk, 2018-03-12:
It's from GIT this morning. I verified it was showing version 0.2 
It also frags the Xagyl_Wheel_config.xml file removing all the filter names.

Blueshawk, 2018-03-12:
I got snowed in so this is my work for today. :dancing_women:

knro, 2018-03-12:
Can you remove Xagy config XML file and try again?

Blueshawk, 2018-03-12:
Yes. SAme result. I looked at the xml file and noticed the names aren't there originally and it populates the saved one so when it failed to get/set the names due to the crash it saved the empty values from default, while saving other params that I changed. I think it's a secondary effect. 
 Since it works on my xubuntu64 notebook(as server) as well as in single connection testing directly connect to my main pc(also xenial) do you think it could be a compiler issue somehow for these arm systems?

Blueshawk, 2018-03-12:
I cleaned everything out and recompiled, getting only indi-asi from ppa and got the same error. Most of it works now, except this pesky error.
Let me know if I can help in any way. :)

knro, 2018-03-13:
You sure you don't have multiple instances of INDI drivers on your system? run

`which indi_moonlite_focus`

Try to run your driver under gdb and see if you can catch any backtrace:

`gdb --eval-command="set follow-fork-mode child" --args indiserver -v indi_xagyl_wheel`

Blueshawk, 2018-03-13:
Moonlite is okay now. I got that all cleaned up, recompiled and now get a clean start from all drivers again. The xagyl crash happens on connect. 
 I'll try to make gdb go when I get home in a couple hours.

Blueshawk, 2018-03-14:
Bit of a setback. After logging a distributed brute force root password attack from chinese ip addresses I realized the odroid wasn't properly configured. They never got in as I have logged attempts at my router up to the time I pulled the plug on it this evening. I ran out of time to get things patched up to move it off their radar tonight(altering ports and improving ssh keygen only security) but it's secure now. I'll try to get that backtrace asap. It's shaping up nicely but wow...weird seems to be the norm. :dango:

knro, 2018-03-15:
Please git pull and sudo make install libindi then turn on Logging in Ekos for Filter Wheels and attach the log even if it crashes.. at least we will know where it crashes.

Blueshawk, 2018-03-15:
Okay I'll get that this evening, as well as a full xml log from indiserver. The ekos log is effected by the driver restart, but maybe I can watch both and see it.

Blueshawk, 2018-03-15:
I connected to the driver a couple times did a couple commands and included both logs. 
This is with ekos running remotely on my office pc calling indiserver on the Odroid-xu4.

[log_18-12-18.txt](https://github.com/indilib/indi/files/1817278/log_18-12-18.txt)
[log_18-06-06.txt](https://github.com/indilib/indi/files/1817279/log_18-06-06.txt)
[indiserver-log.txt](https://github.com/indilib/indi/files/1817297/indiserver-log.txt)

 Everything is basically ubuntu 16.04 (aka debian based) Xubuntu on the PC and Armbian on the Odroid.
I'm considering swapping the O-droid out for an intel based NUC system as a last resort but would rather to fix this since I suspect others will be having the issue at some point, and the Odroid-xu4 is a pretty nifty little SBC. 
I'm going to plug the rig into my intel notebook, update that and recompile and test that now to see if it is also effected. I'll post logs from that in a bit.

Blueshawk, 2018-03-15:
This log is from the Acer intel bay trail notebook updated from PPA recently. It appears to be working and populates the filter parameters. There is a message about a bad file descriptor after I disconnected that might need attention.
[log_18-46-15.txt](https://github.com/indilib/indi/files/1817352/log_18-46-15.txt)
I'll get the indiserver log for this test and edit this comment with it.

knro, 2018-03-16:
Ok is the issue resolved now in GIT? no other problems?

Blueshawk, 2018-03-16:
No. The problem still effects both my ARM processor units. Raspberry pi3 and Odroid-xu4.
The other (intel) tests are for comparison and to rule out other factors.
 Possibly something to do with processor specific compiling.

Blueshawk, 2018-03-19:
AHA! I think I got it. I changed -D CMAKE_BUILD_TYPE=Debug to Release. And poof! Working driver. I got the instructions from INSTALL and never questioned that. I am now questioning that. :D
  Also, for others building from git, and yes I know this seems obvious to our programmers, but if you repeat the process it's a good idea to delete the files in the build directory before you start so it doesn't pick up some old cache or something. 
Problem solved for now. I'll report any other problems if I find any.

Blueshawk, 2018-03-19:
AHA! I think I got it. I changed -D CMAKE_BUILD_TYPE=Debug to Release. And poof! Working driver. I got the instructions from INSTALL and never questioned that. I am now questioning that. :D
Also, for others building from git, and yes I know this seems obvious to our programmers, but if you repeat the process it's a good idea to delete the files in the build directory before you start so it doesn't pick up some old cache or something.
Problem solved for now. I'll report any other problems if I find any.

AND I DID! Sadly, the driver still crashed with stack, only at a different time and event, somewhere after startup while I was getting the rest of them running. This is crazy.

kecsap, 2018-03-20:
Checklist:
1. Check if you did not install accidentally any INDI stuff under /usr/local/. If it was so, delete those files and recompile install the whole INDI (libindi+3rdparty).
2. Before recompiling (and installing) 3rdparty drivers, libindi must be compiled and installed.

Blueshawk, 2018-03-20:
Interesting. I would not have done that but some process may have, especially since indi_gphoto_ccd is installed there on the working unit along with symlinks for the Canon and Nikon derivatives. I'm not running any of those in the current setup but there could be something in there on the rpi3 and Odroid.
I'll check that out when I get home.

Blueshawk, 2018-03-20:
Checked it out, and unfortunately the Odroid is clean. Only one instance of indi drivers found and they are in /usr/bin

Blueshawk, 2018-03-29:
Considering this driver fails on connect and there have been no other reports, I started to suspect my firmware code could be playing a role, despite the fact that it only has the issue on my ARM based units.
With that in mind I went back and cleaned up some loose arrays in hopes it would help but it has not. 
  Just the same I thought I'd include it in the discussion as a possible problem. I have found a few threads discussing stack smashing on arm units being possibly due to a library file but nothing useful to this problem so far.
Heres the code I'm using for the filter wheel:

[FilterWheelZX.ino.txt](https://github.com/indilib/indi/files/1862278/FilterWheelZX.ino.txt)

knro, 2018-03-30:
I made some updates to INDI to fix some memory leaks. Can you compile from GIT and see if it makes any difference?

Blueshawk, 2018-03-30:
Just got the message after midnight. I'll try first thing in the morning!

Blueshawk, 2018-03-30:
Good morning!
Cleared build directories, updated and rebuilt all of main but unfortunately got the same thing. The code I uploaded will run on any Arduino nano ,and probably most Arduino's in general and is for the filter wheel I am using. 
Because it's not an original Xagyl wheel I worry it may be somehow involved in the problem, although it does work very well in Intel systems. There is no interaction beyond sending and receiving usb commands, but I realized that "could" possibly cause a buffer problem if one of my command responses was too long, but any signing of variable types would be handled at the indi driver.

Here's that LDD: lots of libraries. :) 
[lddfromodroid.txt](https://github.com/indilib/indi/files/1863809/lddfromodroid.txt)
There are no *indi* files in /usr/lib on the Odroid, but there is libindidriver.so on the Intel machine. could this be a clue? I found the library in the build directory so I wonder why it wasn't installed?

kecsap, 2018-03-30:
Ah,yeah sorry. Please try:

ls /usr/lib/arm-linux-gnueabihf/*indi*
ls /usr/local/lib/arm-linux-gnueabihf/*indi*

Blueshawk, 2018-03-30:
"If your stuff crash only on a clean arm system, I suspect there will be some arm specific problem somewhere." - I think so too.  I could set up ssh access if you guys want to poke at it directly. Hook up on IRC, email me a keygen etc. as needed. 

Here's whats in those dirs.
ls /usr/lib/arm-linux-gnueabihf/indi
MathPlugins
ls /usr/lib/arm-linux-gnueabihf/indi/MathPlugins/
libindi_SVD_MathPlugin.so

oscope1@Odroidxu4:/usr/lib$ ls /usr/local/lib/arm-linux-gnueabihf/indi
ls: cannot access '/usr/local/lib/arm-linux-gnueabihf/indi': No such file or directory

Blueshawk, 2018-04-07:
Woot I finally nailed it down! An oversized serial response to a command was being mishandled by the driver, only in the arm compiled system with buffer overrun(stack smashing) hack protection. Thus, it never showed up until I installed the first arm based system.
After running round and round and gathering info, researching the weird error message, groveling a bit and even begging for help, I finally had enough info to realize that though it wasn't failing on the Intel system and was on the Odroid, the key clue and common element was that the failure happened consistently on connect. With this in mind I supposed that it must be mishandling the serial command responses from the filter wheel somehow so I started commenting them out one by one until I finally found where I had cheekily change the serial number response to be a long name and version of the emulator. So I shortened that and :woohoo: it worked! 
Thanks again to everyone who helped along the way! And @knro, you might want to put a parse limit on filter wheel serial responses to prevent actual stack smashing from evil minded telescopes. :evil:

closed 2018-04-07 by Blueshawk

knro, 2018-04-08:
I actually started implementing that recently, but it's not yet extended to most drivers yet, but new drivers would limit the response from the serial device.

Blueshawk, 2018-04-08:
It seems so obvious now. That was a tough one with that canary device in the way.