indilib/indi #542: Possible problems on clang 6

jochym, 2018-03-19:
@knro:  I have noticed that the build fails on clang 6 (https://circleci.com/gh/jochym/indi/738) - the bionic default clang version changed recently. For now the build is forced to clang 5 on bionic (the docker image will be rebuild shortly) but I expect problems down the line on OSX. Maybe it would be better to fix it pro-actively? The problem seems to be in the indiccd.cpp file. Shell I prepare additional clang 6 build?

knro, 2018-03-20:
I'll take a look at it thanks! I also asked Csaba to investigate this as he knows a lot more about this than I do.

knro, 2018-03-20:
Ok virtual destructor added now.

jochym, 2018-03-20:
What shell we do with the clang version? Default in bionic is 6 now. I have forced it to be still 5. Remind me why we are building with clang on linux? I know there was a rational reason but I cannot recall what was it.

jochym, 2018-06-23:
@knro : do we need separate clang6 build ?

knro, 2018-06-23:
I don't recall why we're building clang for Linux, maybe better warning/error reporting than the older GCC series? With gcc 8.x, it's on par with Clang now when it comes to warnings so there is no need for it IMO.

jochym, 2018-06-25:
You mean all clang builds? That would speed up the CI considerably (by one cycle at least so 30%). Shell I drop all of them?

knro, 2018-06-25:
Clang on Linux now required. I believe it is used for Mac, so there is that.

jochym, 2018-06-25:
OK. Will do.

closed 2018-06-25 by jochym