indilib/indi #547: Celestron mount : park failure

broadcastyourseb, 2018-03-29:
I don't know why but alt/az position displays on the handcontroller is different of the alt/az displays in kstars.
The indi driver is not using the get_azalt() to get the current park position. So when you press the park button, the driver send the right command but because of the difference in system coordinate, the mount is going in a weird position park.

At power start, the mount is pointing to the celestial pole and alt position is about 90° on the hand controller. I don't remember the az position.

knro, 2018-03-29:
What's the controller version? What mount exactly? logs?

broadcastyourseb, 2018-03-29:
Controller version 4.21 on a celestron advanced GT (CG5)
You have to wait for the log. I'm at the office now.
I checked the log last night. Indi sent the right command as describe in the celestron command manual #bxxxxxxx:yyyyyyyy
I made the math last night. Coordinates match alt/az display in the current park module. The only difference are alt/az coordinates display on my hand controller.

broadcastyourseb, 2018-03-30:
I don't know, if it has something to do with this issue but after some digging in the web, I found this :+1: 

> The Goto Alt-Az and Get Alt-Az menu items have been renamed Goto Axis Postn and Get Axis Postn where Postn is an abbreviation for Position. Since this feature doesn't actually refer to altitude and azimuth when mounted on a wedge, the new designation is more accurate.
[NexStarHandControlVersion4UsersGuide.pdf](https://github.com/indilib/indi/files/1863205/NexStarHandControlVersion4UsersGuide.pdf)

broadcastyourseb, 2018-03-30:
Few more test :
                   INDI                                  NEXSTAR
Lat              47:13:00                           47°13'00"
Lon             358:27:00                         1°33'00"W
Time UTC   2018-03-30T15:44:42      03/30/18 17:51 (Local)
Time offset  2                                      2

Same display on both indi and handcontroller