indilib/indi #556: Feature request: meridian limit setting for EQMOD

StarryBird, 2018-04-06:
Current version of the EQMOD only supports setting horizon limit and the meridian limit setting is not available. When I manually nudge the mount, the mount will not stop slewing even if it passes the meridian, which may damage the mount.  Is there any plan for implementing the feature in the future version? 

Thanks !

StarryBird, 2018-05-23:
@knro Are there any plans for this feature ?  I think it is important to have this feature to protect the mount.

knro, 2018-05-23:
I'm not exactly sure about the current behavior.

@geehalel What do you think?

geehalel, 2018-05-23:
The current behavior is to leave the motor run, be it while slewing or while tracking. I've added Horizon limits after a request some time ago, but I wonder if a driver is the right place to manage such things. At least it may be placed at the telescope level class as this is common to all telescope drivers. On my mind it would be better to leave those aspects to clients themselves.

knro, 2018-05-24:
I agree with Jean-Luc assessment, this belongs to the client.

StarryBird, 2018-05-24:
@knro How about implementing a meridian limit settings in mount control page in ekos for equatorial mount?

When the meridian limit is on and the mount has passed the meridian, stop the mount immediately, however, user can still slew the mount to the opposite direction at this time (user cannot slew the mount along with the tracking direction as the mount has passed the meridian, which is danger) .

How do you think of that ?

knro, 2019-09-18:
Thanks for the report. Please open this issue in the 3rdparty repo issues --> https://github.com/indilib/indi-3rdparty/issues

closed 2019-09-18 by knro