indilib/indi #560: indi-fli make failure

gschwim, 2018-04-10:
Pulled the latest from git, compiled libindi, libfli fine. Moving to indi-fli I get the following.

```
schwim@observatory ~/src/build/indi_3rdparty/indi-fli $ make
[ 33%] Built target indi_fli_wheel
[ 50%] Linking CXX executable indi_fli_ccd
CMakeFiles/indi_fli_ccd.dir/fli_ccd.cpp.o: In function `FLICCD::updateProperties()':
/home/schwim/src/indi/3rdparty/indi-fli/fli_ccd.cpp:154: undefined reference to `INDI::DefaultDevice::SetTimer(int)'
CMakeFiles/indi_fli_ccd.dir/fli_ccd.cpp.o: In function `FLICCD::TimerHit()':
/home/schwim/src/indi/3rdparty/indi-fli/fli_ccd.cpp:681: undefined reference to `INDI::DefaultDevice::SetTimer(int)'
/home/schwim/src/indi/3rdparty/indi-fli/fli_ccd.cpp:689: undefined reference to `INDI::DefaultDevice::SetTimer(int)'
/home/schwim/src/indi/3rdparty/indi-fli/fli_ccd.cpp:796: undefined reference to `INDI::DefaultDevice::SetTimer(int)'
collect2: error: ld returned 1 exit status
CMakeFiles/indi_fli_ccd.dir/build.make:99: recipe for target 'indi_fli_ccd' failed
make[2]: *** [indi_fli_ccd] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/indi_fli_ccd.dir/all' failed
make[1]: *** [CMakeFiles/indi_fli_ccd.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

gschwim, 2018-04-10:
Decided to refresh from the PPA.  The indiserver won't start with indi_fli_ccd:

```
schwim@observatory ~ $ indiserver -vvv indi_fli_ccd
2018-04-10T02:53:38: startup: indiserver -vvv indi_fli_ccd 
2018-04-10T02:53:38: Driver indi_fli_ccd: pid=23157 rfd=3 wfd=6 efd=7
2018-04-10T02:53:38: listening to port 7624 on fd 4
2018-04-10T02:53:38: Driver indi_fli_ccd: sending msg copy 1 nq 1:
<getProperties version='1.7'/>

2018-04-10T02:53:38: Driver indi_fli_ccd: indi_fli_ccd: symbol lookup error: indi_fli_ccd: undefined symbol: _ZN4INDI13DefaultDevice8SetTimerEi
Child process 23157 died
2018-04-10T02:53:38: Driver indi_fli_ccd: stderr EOF
2018-04-10T02:53:38: Driver indi_fli_ccd: restart #1
2018-04-10T02:53:38: Driver indi_fli_ccd: pid=23158 rfd=0 wfd=6 efd=7
2018-04-10T02:53:38: Driver indi_fli_ccd: sending msg copy 1 nq 1:
<getProperties version='1.7'/>
```

knro, 2018-04-10:
I just tried to do a clean compile for everything and couldn't get the error. I also tried to update to 1.7 from PPA on a VM and it worked fine. Most likely you have an old INDI library installation lurking somewhere in /usr/local or another system path. Check it out and remove it.

gschwim, 2018-04-11:
Refreshed from git and working now. Must've overwritten the offender. Thx for the help!

closed 2018-04-11 by gschwim