indilib/indi #573: Wifi DSLR support

vedranf, 2018-04-14:
Hello,

I have Nikon DSLR with wifi support and I'd like to use it with indi without yet another cable if not needed. Given that gphoto supports accessing cameras over wifi, this shouldn't be difficult to implement. In fact, I took a look at the code and made a proof of concept patch which appears to work fine for my case (didn't test much yet). In addition, it would be great to have an option not to download images from the camera because it's slow over wifi, but rather just store them on SD card. I know that for polar alignment, etc one needs to get the images, but after that is done, there is no need to transfer them as post-processing will be done later.

Here's the patch (in my case, camera IP is always the same so I hardcoded it):

--- gphoto_ccd.old.cpp  2018-04-14 12:03:23.983554687 +0200
+++ gphoto_ccd.cpp      2018-04-12 02:15:31.027927001 +0200
@@ -54,6 +54,7 @@
                                     { "indi_canon_ccd", "Canon DSLR", "Canon" },
                                     { "indi_nikon_ccd", "Nikon DSLR", "Nikon" },
                                     { "indi_pentax_ccd", "Pentax DSLR", "Pentax" },
+                                    { "indi_nikon_ccd", "Nikon DSLR", "PTP/IP Camera" },
                                     { NULL, NULL, NULL } };
 
 /**********************************************************
@@ -98,6 +99,8 @@
             const char *model, *port;
             gp_list_reset(list);
             cameraCount = gp_camera_autodetect(list, context);
+            gp_list_append(list, "PTP/IP Camera", "ptpip:192.168.1.1");
+            cameraCount = 1;
             /* Now open all cameras we autodected for usage */
             IDLog("Number of cameras detected: %d.\n", cameraCount);
 

Thanks,
Vedran

knro, 2018-04-22:
Great, please send in a PR!

knro, 2018-04-22:
Though hard coding the IP address is not a good idea. So the IP address must be saved somewhere.. Having it as a property would be challenging since we still do the actual configuration file. Need to think this one more.

knro, 2019-09-18:
Thanks for the report. Please open this issue in the 3rdparty repo issues --> https://github.com/indilib/indi-3rdparty/issues

closed 2019-09-18 by knro