indilib/indi #576: SynScan driver always stays in "slewing" mode even when done slewing (PATCH)

k-ross, 2018-04-21:
When using the SynScan telescope driver, after issuing a GoTo command, the status is always reported as slewing, even when it's done.  This causes several problems, like capture won't work while it thinks the mount is slewing, and neither will the polar alignment assistant.  I'm sure there are other places where it causes problems as well.

Here is my fix for the problem.  It works for me on my setup.  It may or may not work on all mounts supported by the SynScan driver

```
diff --git a/libindi/drivers/telescope/synscanmount.cpp b/libindi/drivers/telescope/synscanmount.cpp
index e50cbd6b..93990a3d 100644
--- a/libindi/drivers/telescope/synscanmount.cpp
+++ b/libindi/drivers/telescope/synscanmount.cpp
@@ -364,14 +364,22 @@ bool SynscanMount::ReadScopeStatus()
     numread = tty_read(PortFD, str, 2, 2, &bytesRead);
     if (str[1] == '#')
     {
-        if ((int)str[0] == 0)
+        TrackingStatus = str[0];
+        switch((int)str[0])
+        {
+        case 0:
             TrackingMode = "Tracking off";
-        if ((int)str[0] == 1)
+            break;
+        case 1:
             TrackingMode = "Alt/Az tracking";
-        if ((int)str[0] == 2)
+            break;
+        case 2:
             TrackingMode = "EQ tracking";
-        if ((int)str[0] == 3)
+            break;
+        case 3:
             TrackingMode = "PEC mode";
+            break;
+        }
     }
 
     UpdateMountInformation(true);
@@ -386,10 +394,12 @@ bool SynscanMount::ReadScopeStatus()
         {
             //  Nothing to do here
         }
-        else
+        else if (MountCode < 128)
         {
-            if (TrackState == SCOPE_PARKING)
-                TrackState = SCOPE_PARKED;
+            if (TrackingStatus[0] != 0)
+                TrackState = SCOPE_TRACKING;
+            else
+                TrackState = SCOPE_IDLE;
         }
     }
     if (TrackState == SCOPE_PARKING)
diff --git a/libindi/drivers/telescope/synscanmount.h b/libindi/drivers/telescope/synscanmount.h
index aaaa3906..37270725 100644
--- a/libindi/drivers/telescope/synscanmount.h
+++ b/libindi/drivers/telescope/synscanmount.h
@@ -98,5 +98,6 @@ class SynscanMount : public INDI::Telescope
     std::string AlignmentStatus;
     std::string GotoStatus;
     std::string MountPointingStatus;
+    std::string TrackingStatus;
     std::string TrackingMode;
 };
```

knro, 2018-04-21:
@kecsap Can you please check this?

kecsap, 2018-04-22:
It seems to be okay.

knro, 2018-04-22:
@k-ross Ok please submit a PR with the change

closed 2018-04-25 by k-ross