indilib/indi #580: Moravian G2-8300FW getting libusb_bulk_transfer failed

sergio-dr, 2018-04-25:
@sWski Hi, I've reported this issue on INDI forum, http://indilib.org/forum/ccds-dslrs/3302-moravian-g2-8300fw-getting-libusb-bulk-transfer-failed.html , please take a look at that if you have a minute.

Since then I made some testing with the example code included in the library (downloaded from http://www.gxccd.com/cat?id=156&lang=409 ) to rule out the indi driver, and I was able to reproduce it (only once), getting the same error from the library. 

The worst part is that issue seems to appear randomly, but I cannot reproduce it on my Windows laptop (using Maxim DL and the Gx driver for 3rd party packages), so I think it shouldn't be a hardware problem on the camera itself. I've tested extensively other USB devices (including an USB3.0  ZWO ASI camera) on the Intel NUC without stability problems.

As far as I tested, this error only appeared when imaging subframes, but obviously I cannot be sure if this makes sense or not. Also, could the "bulk endpoint [...] has invalid maxpacket 64" dmesg messages be related to this?

Any ideas? 
Thanks in advance.
Sergio

sterne-jaeger, 2018-05-05:
I can confirm the problem, exactly the same behavior on Raspberry Pi 3b with Raspbian 9.0 and INDI 1.7.2.

The problem seems to occur during focusing and receiving subframes.
Wolfgang

knro, 2018-05-05:
So if you subframe from capture module (without focusing), can you trigger this error?

sergio-dr, 2018-05-05:
Yes. But it also happens when exposuring the whole frame. I think it's simply that when subframing, downloads from camera are more frequent. 
I've tried to involve Moravian (no luck for now) as I reproduced the error using (a modified version) of the example provided in the gxccd library. 
Thanks for your interest.

Subframe:
```
[2018-04-25T21:22:25.391 CEST DEBG ][   org.kde.kstars.ekos.capture] - Preparing capture job "Light" for execution.
[2018-04-25T21:22:25.403 CEST INFO ][   org.kde.kstars.ekos.capture] - "Changing filter to S2R..."
[2018-04-25T21:22:25.445 CEST INFO ][           org.kde.kstars.indi] - MI CCD G2-8300 :  "[INFO] Setting current filter to slot 3 "
[2018-04-25T21:22:29.759 CEST INFO ][   org.kde.kstars.ekos.capture] - "Capturing image..."
[2018-04-25T21:22:29.761 CEST DEBG ][           org.kde.kstars.indi] - MI CCD G2-8300 : "[DEBUG] Filter changed to 3 "
[2018-04-25T21:22:29.800 CEST DEBG ][           org.kde.kstars.indi] - MI CCD G2-8300 : "[DEBUG] Requested CCD Frame is (500,765) (135 x 145) "
[2018-04-25T21:22:29.844 CEST DEBG ][           org.kde.kstars.indi] - MI CCD G2-8300 : "[DEBUG] The Final image area is (500, 765), (635, 910) "
[2018-04-25T21:22:31.503 CEST DEBG ][           org.kde.kstars.indi] - MI CCD G2-8300 : "[DEBUG] Taking a 1.000 seconds frame... "
[2018-04-25T21:22:31.505 CEST DEBG ][           org.kde.kstars.indi] - MI CCD G2-8300 : "[DEBUG] Exposure in progress: Time left 1.00s "
[2018-04-25T21:22:34.646 CEST DEBG ][           org.kde.kstars.indi] - MI CCD G2-8300 : "[DEBUG] Uploading file. Ext: fits, Size: 43200, sendImage? Yes, saveImage? No "
[2018-04-25T21:22:34.663 CEST INFO ][           org.kde.kstars.indi] - "FITS" file saved to "/home/aristarco/datos/Light/S2R/Light_031.fits"
[2018-04-25T21:22:34.664 CEST DEBG ][           org.kde.kstars.indi] - MI CCD G2-8300 : "[DEBUG] Upload complete "
[2018-04-25T21:22:34.665 CEST WARN ][        org.kde.knotifications] - Audio notification requested, but sound file from notifyrc file was not found, aborting audio notification
[2018-04-25T21:22:34.665 CEST INFO ][           org.kde.kstars.fits] - Loading FITS file  "/home/aristarco/datos/Light/S2R/Light_031.fits"
[2018-04-25T21:22:34.673 CEST INFO ][           org.kde.kstars.fits] - Loading FITS file  "/home/aristarco/datos/Light/S2R/Light_031.fits"
[2018-04-25T21:22:34.697 CEST INFO ][   org.kde.kstars.ekos.capture] - "Received image 1 out of 200."
...
[2018-04-25T21:24:43.298 CEST INFO ][   org.kde.kstars.ekos.capture] - "Received image 32 out of 200."
[2018-04-25T21:24:43.313 CEST INFO ][   org.kde.kstars.ekos.capture] - "Capturing image..."
[2018-04-25T21:24:44.254 CEST DEBG ][           org.kde.kstars.indi] - MI CCD G2-8300 : "[DEBUG] Taking a 1.000 seconds frame... "
[2018-04-25T21:24:44.259 CEST DEBG ][           org.kde.kstars.indi] - MI CCD G2-8300 : "[DEBUG] Exposure in progress: Time left 1.00s "
[2018-04-25T21:25:45.392 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2018-04-25T19:25:45: Driver indi_mi_ccd_usb: [gxccd] error: libusb_bulk_transfer failed(): cmd: 0x11, write: false, ret: -7, Operation timed out, transferred: 0, in_len: 2, out_len: 5"
[2018-04-25T21:25:45.392 CEST DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2018-04-25T21:25:45.393 CEST INFO ][           org.kde.kstars.indi] - MI CCD G2-8300 :  "[ERROR] Getting voltage failed: Camera is not connected. "
[2018-04-25T21:25:45.395 CEST INFO ][           org.kde.kstars.indi] - MI CCD G2-8300 :  "[ERROR] Getting image ready failed: Camera is not connected. "
[2018-04-25T21:25:46.396 CEST INFO ][           org.kde.kstars.indi] - MI CCD G2-8300 :  "[ERROR] Getting temperature failed: Camera is not connected. "
[2018-04-25T21:25:46.398 CEST INFO ][           org.kde.kstars.indi] - MI CCD G2-8300 :  "[ERROR] Getting voltage failed: Camera is not connected. "
[2018-04-25T21:25:46.401 CEST INFO ][           org.kde.kstars.indi] - MI CCD G2-8300 :  "[ERROR] Getting image ready failed: Camera is not connected. "
[2018-04-25T21:25:47.396 CEST INFO ][           org.kde.kstars.indi] - MI CCD G2-8300 :  "[ERROR] Getting temperature failed: Camera is not connected. "
[2018-04-25T21:25:47.398 CEST INFO ][           org.kde.kstars.indi] - MI CCD G2-8300 :  "[ERROR] Getting voltage failed: Camera is not connected. "
```

Full frame (dark):
```
[2018-05-01T14:37:59.292 CEST DEBG ][   org.kde.kstars.ekos.capture] - Preparing capture job "Dark_300_secs" for execution.
[2018-05-01T14:37:59.300 CEST INFO ][   org.kde.kstars.ekos.capture] - "Setting temperature to -20 C..."
[2018-05-01T14:38:20.188 CEST INFO ][   org.kde.kstars.ekos.capture] - "Capturing image..."
[2018-05-01T14:38:21.039 CEST DEBG ][           org.kde.kstars.indi] - MI CCD G2-8300 : "[DEBUG] Taking a 300.000 seconds frame... "
[2018-05-01T14:38:21.300 CEST DEBG ][           org.kde.kstars.indi] - MI CCD G2-8300 : "[DEBUG] Exposure in progress: Time left 299.74s "
[2018-05-01T14:38:22.413 CEST DEBG ][           org.kde.kstars.indi] - MI CCD G2-8300 : "[DEBUG] Exposure in progress: Time left 298.63s "
[2018-05-01T14:38:23.525 CEST DEBG ][           org.kde.kstars.indi] - MI CCD G2-8300 : "[DEBUG] Exposure in progress: Time left 297.51s "
[2018-05-01T14:38:24.638 CEST DEBG ][           org.kde.kstars.indi] - MI CCD G2-8300 : "[DEBUG] Exposure in progress: Time left 296.40s "
...
[2018-05-01T14:43:20.667 CEST DEBG ][           org.kde.kstars.indi] - MI CCD G2-8300 : "[DEBUG] Exposure in progress: Time left 0.37s "
[... cut ...]
[2018-05-01T14:44:51.821 CEST INFO ][           org.kde.kstars.indi] - MI CCD G2-8300 :  "[ERROR] Getting image ready failed: Camera is not connected. "
[2018-05-01T14:44:52.926 CEST INFO ][           org.kde.kstars.indi] - MI CCD G2-8300 :  "[ERROR] Getting temperature failed: Camera is not connected. "
[2018-05-01T14:44:52.930 CEST INFO ][           org.kde.kstars.indi] - MI CCD G2-8300 :  "[ERROR] Getting voltage failed: Camera is not connected. "
[2018-05-01T14:44:52.934 CEST INFO ][           org.kde.kstars.indi] - MI CCD G2-8300 :  "[ERROR] Getting image ready failed: Camera is not connected. "
```

sWski, 2018-05-05:
Hello guys, thank you for your reports.

@sergio-dr I wrote you an email on wednesday, maybe it wasn't delivered.

I think that I've successfully reproduced this behavior yesterday and it seems as a deadlock issue. I'll test it next week and will push fixed library.

sergio-dr, 2018-05-05:
@sWski Thank you very much! That's great news indeed. This was making me nuts. I'll gladly test it when the new version of the library is ready. 
For some reason gmail labeled your email as spam so I didn't see it. Thanks for answering.

sWski, 2018-05-09:
Dear @sergio-dr,
do you know how to build the indi_mi_ccd driver from command line? I've prepared fixed library (I suppose you have 64-bit OS) [here](http://www.gxccd.com/download/libgxccd64.tar.gz). You have to overwrite the file in your indi_mi_ccd directory and rebuild the driver itself. 

Please let me know whether it helped or not.

sergio-dr, 2018-05-09:
Hi @sWski , yes I'm building indi from git. I'll test as soon as I can. Can you suggest any test case where the error was more likely to occur (with the previous version of the library, I mean), please? Sometimes it was difficult to reproduce the error, even subframing to a small area in order to get more frequent downloads from the camera, and reducing the polling period in INDI driver (used when querying ccd temperature and cooling power). 
Thanks for getting involved.

sWski, 2018-05-09:
I have extended our example in the libgxccd library with second thread which was polling for cooling power every 10ms. Reducing polling period and subframing small area should definitely lead to this error.

sterne-jaeger, 2018-05-09:
Dear Jakub,
could you please provide a 32bit version as well? 
I am running INDI on a Raspberry PI 3b with Raspbian 9 (stretch).
Wolfgang

sWski, 2018-05-10:
Hello Wolfgang,
I've prepared archive which contains all versions of our library and you can download it [here](http://www.gxccd.com/download/libgxccdbin.tar.gz).

sergio-dr, 2018-05-18:
Sorry for the late report, last Friday night I had a typical field session involving focusing on several filters, framing, doing multiple light and calibration exposures, etc. spanning 6 hours, and everything was fine. But I, unlike Wolfgang, am on a x86_64 platform (Intel NUC, Ubuntu Mate 18.04). 
Wolfgang, sorry to hear that. I'll keep using kstars/ekos in further sessions to see if the problem arises again, and report back. Did you get the '[gxccd] error: libusb_bulk_transfer failed()' messages?

sterne-jaeger, 2018-05-18:
Hi sergio-dr,
no, I did not see it - at least not in the INDI log. Which log did you mean?
But today everything is running fine. So let's wait and see...

sergio-dr, 2018-05-19:
I captured those libusb errors from the kstars log file (I didn't look at the INDI logs, thought that those error messages should be also there).

knro, 2018-05-20:
Maybe increase the usbfs memory on Rpi3?

echo 256 >/sys/module/usbcore/parameters/usbfs_memory_mb

This is used in ASI cameras, I imagine it would help with MI as well.

sterne-jaeger, 2018-05-20:
Good point. I'll give it a try.

sterne-jaeger, 2018-05-21:
Last night everything went fine, so hopefully the bug has gone. I updated the udev rules for MI cameras in /lib/udev/rules.d/99-miccd.rules:

`ACTION=="add", ATTR{idVendor}=="1347", RUN+="/bin/sh -c '/bin/echo 256 >/sys/module/usbcore/parameters/usbfs_memory_mb'"`

sWski, 2018-05-21:
I'll open PR with new libraries. If you run into the bug again, feel free to contact me.

@knro Should I update that udev rule for all users or add it to README only?

PR #601 (2018-05-21): ⬆️ Update library to version 0.3.4
Hopefully fixes #580.
closed 2018-05-21 by knro