indilib/indi #581: SynScan on macOS 10.13.4 not working

toxik, 2018-04-26:
Hello, 

I'm using KStars 2.9.4 and trying to connect to an SynScan EQ5 Pro via an USB to RS232 adapter (Prolific Technology, Inc.) connected to the handset.

I've tried both with and without PC Direct Mode but no luck, this is what I always get:
```
org.kde.kstars.indi: SynScan :  "[ERROR] Failed to connect to port (/dev/cu.usbserial). Error: Port failure Error: Resource busy. Check if device is connected to this port. "
```

The mount works nicely with EQMac v2.0.0 beta3 in PC Direct Mode with the same setup.

Any thoughts ?

toxik, 2018-04-26:
I tried with the EQMod driver as well, I get the same "resource busy" error. Any way I could debug this further?

knro, 2018-04-26:
Most likely then an issue with something holding the driver resources in Mac. Make sure everything is closed and nothing is using the serial port for anything.

kecsap, 2018-04-26:
Is the device node the same in EQMac and KStars? /dev/cu.usbserial?

toxik, 2018-04-26:
hmm not sure, this is what EQMac says:

[image]

toxik, 2018-04-26:
OK I just rebooted everything one more time, started all fresh and it worked with the SynScan driver.
Sorry for the wasted time !

closed 2018-04-26 by toxik