indilib/indi #591: GPSNMEA GPS disconnect behavior

deisenlord, 2018-05-11:
I've discovered if I let my iphone fall asleep or otherwise shutdown the GPS app on my phone the indi gpsnmea driver doesn't handle it gracefully.  The result is one or more of kstars, indi-serrver, indi_gpsnmea pinning a core or two at 100%.   The result is not recoverable, on a stellarmate memory will quickly be exhausted and things will crash, on OSX things become non-responsive and you need to kill everything.

The reason appears to be two fold
a). tty_nread_section doesn't really support the concept of EOF or reads of zero bytes.
2). When my IPhone GPS device is shutdown, tty_read_nsection doesn't receive errors or timeouts but just encounters reads of zero bytes (traditional EOF).

tty_rneadsection doesn't handle reads of zero bytes as a problem, it just moves its buffer pointer ahead and marches on.   If the input buffer wasn't cleared before the call, then it will just march though the old buffer contents until it sees a LF and returns the same result as the last valid call, over and over again.
This puts the thread in the driver into a tight spin loop. If you clear the buffer ahead of time, the zero length reads will  eventually result in an overflow.  I used this to show how I'd like indi_gpsnmea to behave if the GPS device disconnects, and that is for the driver to disconnect as well.   Clearly my little fragment below is not how this should be accomplished.
`
        memset(line, '\0', MINMEA_MAX_LENGTH) ;
        int tty_rc = tty_nread_section(PortFD, line, MINMEA_MAX_LENGTH, 0xA, 3, &bytes_read);
        if (tty_rc < 0)
        {
            if (tty_rc == TTY_OVERFLOW)
            {
                LOG_WARN("Appears GPS has disconnected so we will disconnect as well") ;
                INDI::GPS::setConnected(false) ;
                updateProperties() ;
                break ;
`
            }

I'm guessing this is a trival fix for someone with good knowledge of indicon.c
If there is a robust way to "test" the connection before the tty_nread_section call I think this would be the ticket.  

Any advice would be appreciated.
Thanks!
David

closed 2018-05-11 by knro

knro, 2018-05-11:
Thanks for the report & fix. Please git pull and let me know if it now works.

deisenlord, 2018-05-11:
Hi @knro, this won't work without the memset before the tty_nread_section, otherwise it won't overflow and just return the old info from the buffer.

commit 2ab23316 (2018-05-11): Clear buffer in tty_nread_Section. Fixes #591
commit 44aaf5d3 (2018-05-11): Fix INDI GPS NMEA driver as per David Eisenlord suggestions. Fixes #591