indilib/indi #593: FLI driver failed to compile

lluiscanals, 2018-05-11:
FLI driver versions 1.3 and 1.3.2 fail to compile in kernels >= 4.15.X (for Ubuntu 18). The following error is thrown:

`make[1]: Entering directory '/usr/src/linux-headers-4.15.0-20-generic'
  CC [M]  /home/indi/indi/src/INDI/fli/fliusb/fliusb.o
/home/indi/indi/src/INDI/fli/fliusb/fliusb.c: In function ‘fliusb_sg_bulk_read’:
/home/indi/indi/src/INDI/fli/fliusb/fliusb.c:381:19: error: ‘struct timer_list’ has no member named ‘data’
   dev->usbsg.timer.data = (unsigned long)dev;
                   ^
/home/indi/indi/src/INDI/fli/fliusb/fliusb.c:382:29: error: assignment from incompatible pointer type [-Werror=incompatible-pointer-types]
   dev->usbsg.timer.function = fliusb_sg_bulk_read_timeout;
                             ^
/home/indi/indi/src/INDI/fli/fliusb/fliusb.c: In function ‘fliusb_initdev’:
/home/indi/indi/src/INDI/fli/fliusb/fliusb.c:817:3: error: implicit declaration of function ‘init_timer’; did you mean ‘init_timers’? [-Werror=implicit-function-declaration]
   init_timer(&tmpdev->usbsg.timer);
   ^~~~~~~~~~
   init_timers
cc1: some warnings being treated as errors
scripts/Makefile.build:339: recipe for target '/home/indi/indi/src/INDI/fli/fliusb/fliusb.o' failed`

It seems that the 'init_timer' function has been changed/modified for kernels 4.15.X to: Use timer_setup() macro and new timeout function definition.

knro, 2018-05-11:
The INDI FLI driver _no_ longer uses the FLI USB kernel driver, it's obsolete. The updated INDI driver now uses an updated version of FLI SDK that works with libusb.

closed 2018-05-11 by knro

lluiscanals, 2018-05-11:
Perfect, I will give it a try! Thanks!