indilib/indi #617: Support for bulb capture over SNAP Port

rootdarkarchon, 2018-06-11:
Hello,

the original EQMOD allows bulb capture for cameras that don't have BULB over USB utilizing the SNAP port on the mount (EQ6-R, AZEQ6). Would it be possible to implement this somehow so I could utilize shooting bulb with my D5100? Currently I am running a Windows based setup with software where I can set up bulb over telescope mount and download the images separately. I would love to switch to a Linux based approach, this USB issue is one of the reasons I cannot.

A suggestion would be exposing the SNAP port functionality as a ttyUSB device so it can act as a fake serial shutter, so the integration with GPhoto DSLR functionality will be easy.

vdelord, 2018-10-10:
Hi, I just bought an EQ6-R too and being able to trigger my camera sequentially through the mount and snap port would be really nice. On my setup I don't download the photos until my shooting session is done. It would be nice to remove the need of a link between my raspberry and the camera.

geehalel, 2018-12-10:
Hi,
I've implemented snap port control some months ago, it is on my [repo](https://github.com/geehalel/indi) in the eqsnap branch. I've just resynced it with the current libindi repo so you may test it with your mount.  There is a small python script in the repo (`eqsnap1.py`) to launch at startup which will allow the gphoto driver to control the snap port as if it was a serial port (`/tmp/eqsnapport1`). 
I would need the mount code for the EQ6-R as snap port control is only activated for EQ8, AZ-EQ6 and AZ-EQ5 (2 ports for the AZ-EQ5, not sure for others).

Blueshawk, 2018-12-23:
Woot. I did a search after getting an EQ6-r myself, after realizing the snap port might be useful for my non tethering k-50 and found this. Looks like a good start. :D I think I saw the protocol around somewhere. Hopefully somebody will find it soon. If not, sniffing the serial line to the mount may show the needed command to trigger it.

Blueshawk, 2019-01-07:
I noticed a shutter port in indi_pentax_ccd(and I think maybe gphoto_ccd?) Would this be the other end of that train? My eq6r reports in indi status as an azeq6, making me think the same code from that driver might be worth a try. Aren't most of them using synscan protocol? My eq6r came with a synscan control pad but I'm using a direct serial-usb cable now. I might tinker with this if I get time but my plate is pretty full right now.

geehalel, 2019-01-07:
Yes the shutter port comes from gphoto_ccd and is the other end to use. When using indi_eqmod snap port it will be a virtual port managed by the python script I referenced above.
As the EQ6-R reports as the AZ-EQ6 you may have a try if you want. I'll make a pull request after someone tests it. Thanks.

knro, 2019-03-05:
Hello all, how is this feature coming along? Was it tested by other in EQMod with @geehalel branch?

knro, 2019-09-18:
Thanks for the report. Please open this issue in the 3rdparty repo issues --> https://github.com/indilib/indi-3rdparty/issues

closed 2019-09-18 by knro