indilib/indi #619: MAKE fails with undefined reference to curl_easy and curl_global

justanenduser, 2018-06-16:
CMAKE finds the CURL library just fine
	curl/libcurl/libcurl-devel ver 7.59.0-4

MAKE fails with - 
Linking CXX executable <driver name>
libindidriver.so.1.7.3: undefined reference to `curl_easy_perform'
libindidriver.so.1.7.3: undefined reference to `curl_easy_init'
libindidriver.so.1.7.3: undefined reference to `curl_easy_cleanup'
libindidriver.so.1.7.3: undefined reference to `curl_easy_setopt'
libindidriver.so.1.7.3: undefined reference to `curl_global_init'
libindidriver.so.1.7.3: undefined reference to `curl_global_cleanup'
collect2: error: ld returned 1 exit status

I'm running an up2date (a/o 06/16/2018) Fedora 28, kernel 4.16.15-300, with gcc 8.1.1 and cmake 3.11.2 -

gcc version 8.1.1 20180502 (Red Hat 8.1.1-1) (GCC)
Using built-in specs.
COLLECT_GCC=/usr/bin/gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/8/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-redhat-linux
Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,fortran,objc,obj-c++,ada,go,lto --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --with-linker-hash-style=gnu --enable-plugin --enable-initfini-array --with-isl --enable-libmpx --enable-offload-targets=nvptx-none --without-cuda-driver --enable-gnu-indirect-function --enable-cet --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux
Thread model: posix

Adding -lcurl to the end of each link.txt file will allow the subsequent reissue of MAKE to succeed. 

Please advise ... :-)

knro, 2018-06-17:
Curl is required for only two drivers, not the whole library. Is this failing at a specific driver?

justanenduser, 2018-06-17:
Linking fails at any driver (filter-wheel, focus, telescope, weather, gps,
dome, rotator, etc) until I add -lcurl to their respective link.txt file.
This happens at the first stage, inside the libindi subdirectory, and at
the second stage, inside the 3rdparty subdirectory. I have thus far failed
at my attempts to understand the basics of cmake. At the moment I'm not
sure where I should be digging, into cmake or my Fedora
distribution/configuration.

lupinix, 2018-06-18:
Cannot reproduce here (F27 and F28), also the builds in my repository are fine (check .spec file for cmake call and BuildRequires): https://build.opensuse.org/package/show/home:lupinix-indi:fedora-bleeding/libindi-bleeding

justanenduser, 2018-06-19:
Thanks for your efforts ... you are correct. I have two machines (desktop and laptop) that I believe to be identically configured ... INDI builds and installs as expected on the laptop. The error I reported occurs on the desktop. So far I have been unable to find the difference ... sigh. We can close this report.

closed 2018-06-19 by justanenduser

z3ntu, 2018-09-13:
I am experiencing the same issue when building libindi as a flatpak module.
Build log:
```
========================================================================
Building module indi in /mnt/hdd/flathub/.flatpak-builder/build/indi-7
========================================================================
-- The C compiler identification is GNU 6.2.0
-- The CXX compiler identification is GNU 6.2.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPATIBLE_FORTIFY_SOURCE
-- Performing Test COMPATIBLE_FORTIFY_SOURCE - Success
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found ZLIB: /usr/lib/libz.so (found version "1.2.8") 
-- Found CFITSIO 3.45: /app/lib/libcfitsio.a
-- Found CFITSIO: /app/lib/libcfitsio.a
-- Found NOVA: /app/lib/libnova.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Found USB1: /app/lib/libusb-1.0.so (found version "1.0.22") 
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME - Success
-- Found CURL: /usr/lib/libcurl.so (found version "7.50.1") 
-- Found GSL: /app/include (found version "2.5") 
-- Found JPEG: /usr/lib/libjpeg.so  
-- Found OGGTHEORA: /usr/lib/libogg.so  
-- Could NOT find GTest (missing:  GTEST_LIBRARY GTEST_INCLUDE_DIR GTEST_MAIN_LIBRARY) 
-- Could NOT find GMock (missing:  GMOCK_LIBRARY GMOCK_INCLUDE_DIR) 
-- GTEST not found, not building unit tests
-- 
-- The following OPTIONAL packages have been found:

 * PkgConfig
 * OggTheora

-- The following REQUIRED packages have been found:

 * Threads
 * ZLIB
 * CFITSIO , A library for reading and writing data files in FITS (Flexible Image Transport System) data format , <http://heasarc.gsfc.nasa.gov/fitsio/fitsio.html>
   Provides INDI with FITS I/O support.
 * Nova , A general purpose, double precision, Celestial Mechanics, Astrometry and Astrodynamics library , <http://libnova.sourceforge.net>
   Provides INDI with astrodynamics library.
 * USB1
 * CURL
 * GSL
 * JPEG

-- The following OPTIONAL packages have not been found:

 * GTest
 * GMock

[... 62 lines cut ...]
[ 14%] Building CXX object CMakeFiles/indidriver.dir/libs/stream/encoder/rawencoder.cpp.o
[ 15%] Building CXX object CMakeFiles/indidriver.dir/libs/stream/encoder/mjpegencoder.cpp.o
[ 15%] Building CXX object CMakeFiles/indidriver.dir/libs/stream/recorder/theorarecorder.cpp.o
[ 15%] Building C object CMakeFiles/indidriver.dir/libs/webcam/v4l2_colorspace.c.o
[ 16%] Building CXX object CMakeFiles/indidriver.dir/libs/webcam/v4l2_base.cpp.o
[ 16%] Building CXX object CMakeFiles/indidriver.dir/libs/webcam/v4l2_decode/v4l2_decode.cpp.o
[ 16%] Building CXX object CMakeFiles/indidriver.dir/libs/webcam/v4l2_decode/v4l2_builtin_decoder.cpp.o
[ 17%] Building C object CMakeFiles/indidriver.dir/libs/indibase/hid_libusb.c.o
[ 17%] Linking CXX shared library libindidriver.so
[ 17%] Built target indidriver
Scanning dependencies of target indi_meta_weather
[ 18%] Building CXX object CMakeFiles/indi_meta_weather.dir/drivers/weather/weathermeta.cpp.o
[ 18%] Linking CXX executable indi_meta_weather
libindidriver.so.1.7.4: undefined reference to `curl_easy_perform'
libindidriver.so.1.7.4: undefined reference to `curl_easy_init'
libindidriver.so.1.7.4: undefined reference to `curl_easy_cleanup'
libindidriver.so.1.7.4: undefined reference to `curl_easy_setopt'
libindidriver.so.1.7.4: undefined reference to `curl_global_init'
libindidriver.so.1.7.4: undefined reference to `curl_global_cleanup'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/indi_meta_weather.dir/build.make:105: indi_meta_weather] Error 1
make[1]: *** [CMakeFiles/Makefile2:142: CMakeFiles/indi_meta_weather.dir/all] Error 2
make: *** [Makefile:128: all] Error 2
Error: module indi: Child process exited with code 2
```

z3ntu, 2018-09-13:
Apparently `lib/libcfitsio.a` contains the references to `curl_whatever`...

z3ntu, 2018-09-13:
Hmm when there's a libcfitsio shared library (.so), then the issue doesn't happen
```
-rw-r--r-- 2 luca luca 5690012  1. Jän 1970  build-dir/files/lib/libcfitsio.a
lrwxrwxrwx 1 luca luca      15  1. Jän 1970  build-dir/files/lib/libcfitsio.so -> libcfitsio.so.7*
lrwxrwxrwx 1 luca luca      20  1. Jän 1970  build-dir/files/lib/libcfitsio.so.7 -> libcfitsio.so.7.3.45*
-rwxr-xr-x 2 luca luca 1552184  1. Jän 1970  build-dir/files/lib/libcfitsio.so.7.3.45*
```