indilib/indi #634: Telescope wizard/Telescope model

Lagroxane, 2018-06-29:
I'm not sure what is intended here.

Should the user select the type of mount ? Or should the user select their telescope.

If it should be the mount, it might be good to change the window to say "Telescope Mount" or just "Mount." As you can see it allowed me to select my telescope.
[TelWizTelModel.pdf](https://github.com/indilib/indi/files/2151021/TelWizTelModel.pdf)

knro, 2018-06-29:
Please ignore Telescope Wizard and just use Ekos directly. In fact, I will disable Telescope Wizard in the next KStars release. Having said that, this issue tracking is for **INDI** issues (drivers), and not for **client** issues. To report issues with KStars, please file them at bugs.kde.org

closed 2018-06-29 by knro

Lagroxane, 2018-06-29:
Hi Jasem,

I wasn’t sure if I was posting in the right place. Sorry

c