indilib/indi #642: More ASi78mm vs. ekos issues.

Blueshawk, 2018-07-08:
@knro Following up on the asi vs. ekos forum post.
I updated kstars  asi-common and indi-asi to nightly...note I have not done indi-full because 
Gain is working now but I still have a couple more weird problems with the asi178mm. If I set binning to anything I get this when I capture. 

2018-07-08T20:59:58: [ERROR] ASIGetDataAfterExp (1548x1040 #1 channels) error (13) 
2018-07-08T20:59:58: [INFO] Exposure done, downloading image... 
2018-07-08T20:59:57: [ERROR] ASISetROIFormat (1548x1040 @ 2) error (8) 
2018-07-08T20:59:52: [INFO] Exposure done, downloading image... 
2018-07-08T20:58:43: [INFO] Device configuration applied. 
2018-07-08T20:58:43: [INFO] Upload settings set to client only. 
2018-07-08T20:58:43: [INFO] Loading device configuration... 2018-07-08T20:59:42: [INFO] Exposure done, downloading image...   
2018-07-08T20:58:43: [INFO] The CCD Temperature is 31.300000 
2018-07-08T20:58:43: [INFO] CCD is online. Retrieving basic data. 
2018-07-08T20:58:43: [INFO] Setting intital bandwidth to AUTO on connection.  <<--- is this supposed to do this?

Blueshawk, 2018-07-08:
Follow up on my follow up. I moved the sliders for binning in the control panel and the numbers did not change. I then moved the numbers by the arrows and set that. And now I can change the binning again. Something got out of sync somehow? Perhaps an XML file problem? Weird as usual...it's why I test. I find weird things. More as I get it. :)

Blueshawk, 2018-07-09:
Narrowed it a bit more. It appears to only error when hardware bin is set to on(1). Jupiter was so watery eyed over the whole ordeal it ran and hid behind a tree.

knro, 2018-10-22:
Do you see this on the latest GIT?

knro, 2018-10-24:
If you have teamviewer, email me the details and I'll login and check it out.

Blueshawk, 2018-10-24:
That'd be neat! I could get teamviewer installed for that. oh hrm...my DSL .5meg up has been been problematic for vnc video. :/ The price I pay for dark skies. There's also my daytime work schedule vs. our time zones.(at work now)
 Maybe we could do an ssh directly to the mount computer? I could set it up with the camera in the shop and I send you a pubkey and IP and just leave it running with the camera attached. Since it doesn't have a desktop running it should work better over a slow connection.

knro, 2018-10-25:
SSH would work fine as well.

Blueshawk, 2018-10-25:
okay I'll set that up and mail you the particulars when I get it ready.

Blueshawk, 2018-10-26:
Mailed particulars to your ikarus, I hope that's the right place.

knro, 2018-10-26:
Actually, I was able to replicate this issue here, so I asked ZWO to clarify something now.

Blueshawk, 2018-10-26:
LOL. Mission ...complete.

knro, 2019-09-16:
Please reopen if you still have the same issue with latest drivers.

closed 2019-09-16 by knro

Blueshawk, 2019-09-16:
I think it's pretty well sorted now. I've not had troubles with switching binning lately.