indilib/indi #645: incorrect guiding rate set for Ioptron CEM60

fmozza, 2018-07-09:
1. The CEM60 supports two separate guiding rates for RA and DEC respectively.  The IEQ driver supporting the CEM60 only supports one.

2. When I attempt to set the guiding rate to 0.8, the mount controller show the new rate is 0.08.

3. I reached out to iOptron regarding their RS232 protocol to communicate with the mount, indicating the 2014 document only references a single guiding rate.  I got this as a response:

Command: “:AG#”

Response: “nnnn#”

This command gets the right ascension guiding rate and the declination guiding rate. The first 2 “nn” stands for the right ascension guiding rate 0.nn * sidereal rate. The last 2 “nn” stands for the declination guiding rate 0.nn * sidereal rate.

Note: Valid range of guiding rate of right ascension is [0.01, 0.90] * sidereal rate, and valid range of guiding rate of declination is [0.10, 0.99] * sidereal rate. This command only available in equatorial mounts.

Command: “:RGnnnn#”

Response: “1”

This command sets the right ascension guiding rate and the declination guiding rate. The first 2 “nn” stands for the right ascension guiding rate 0.nn * sidereal rate. The last 2 “nn” stands for the declination guiding rate 0.nn * sidereal rate.

Note: Valid range of guiding rate of right ascension is [0.01, 0.90] * sidereal rate, and valid range of guiding rate of declination is [0.10, 0.99] * sidereal rate. This command only available in equatorial mounts.

Hope this helps!

jmh

knro, 2018-07-09:
Thanks for the detailed info. If you enable mount logging, what do you see the response for AG# from your mount?

knro, 2018-07-09:
Actually, that means you're running new firmware. Check this page: http://indilib.org/devices/telescopes/ioptron.html

You should be using iOptronV3 driver.

closed 2018-07-09 by knro

fmozza, 2018-07-10:
Well, that's interesting... my controller and INDI control panel both report the same versions of firmware as 161101 and 170322 for the drives and mainboards.  I tried the iOptronV3 driver - lots of communcation errors, incomplete field (no firmware version reported, and nothing about RA and DEC guide rates.  

I'll ask iOptron support folks what I should have running on this mount.

jmh

fmozza, 2018-07-10:
Jasem,

According to Ioptron support, I have the latest firmware releases.  There is a 170505 for the RA, but they didn't too sure that it is officially out the door as a stable version.

jmh

commit cffa59e8 (2018-07-10): Increase guide rate by a factor of 10.
Should fix #645