indilib/indi #655: Opening a dialog crashes to desktop.

Blueshawk, 2018-07-19:
Kstars/ekos/indi crashes to desktop when I open a dialog in the control panel(set directory), a preview window(save), or in mount model(save a mount model star layout) in Xubuntu 16.04lts I'm using xfce4 desktop, and I'm pretty sure I'm using pcmanfm for volume management on that computer. I have an idea it may be volume management related because according to @knro it doesn't show up on all systems. running kstars from a terminal may cough up a clue. I'll try to remember to do that when I set up next time.

knro, 2018-07-19:
This is not an INDI issue. For KStars bugs, please file them at https://bugs.kde.org

Also, try to test this under Mate or KDE and see if you still have the same error.

closed 2018-07-19 by knro

Blueshawk, 2018-07-19:
oh..okay. I'm on it. I thought it might be but like to sneak up on these things rather than going straight for the biggest one.