indilib/indi #659: QHY9 fails with upgrade to Kstars/indi

fmozza, 2018-07-26:
After upgrade to latest version of Kstar/INDI, I cannot capture a preview image in EKOS from QHY9.  Debug log file attached.
[qhy-issue.log](https://github.com/indilib/indi/files/2229979/qhy-issue.log)

fmozza, 2018-09-20:
did some more testing ... 
QHY9 under Kstars with indi-1.7.2 works.
QHY9 under Kstars with indi-1.7.4 does not -  fails to capture image.  Error message says streaming is active.
I'm willing to help with more testing.  Let me know how I can help.

knro, 2018-09-24:
I will check it tonight and see if I can replicate it. This is from Stable PPA, correct?

fmozza, 2018-09-24:
Yes.

knro, 2018-09-30:
Can you please try nightly PPA with the new QHY SDK?

fmozza, 2018-09-30:
I updated to the nightly PPA - there was no build for Raspberry PI so pulled indi from github. 

I first ran a test with the new kstars/ekos with indi-1.7.4 and the problem was still there.  

I built and installed indi from the github bits, including the qhy driver, and no more problem - I was able to capture a frame.

Thanks for your work on this!  BTW, I like the changes for cooling in the CCD tab, as well as the changes in the guiding tab - nice work!

knro, 2018-10-26:
I believe this was fixed with the latest QHYSDK 3.1 and INDI QHY driver?

fmozza, 2018-10-27:
My QHY9 died a couple of weeks ago, so I am not able to confirm.  If I'm able to fix the camera, I'll verify.

Thanks for your work on this.

jmh

closed 2018-10-27 by fmozza