indilib/indi #663: FocusLynx Clears focus Position on Connect

AstroMuck, 2018-07-28:
When I reconnect to FocusBoss II/FocusLynx the focus position is set to 0 rather than pulling the last focus position from the FocusBoss.  This forces you to document the final focus position prior to disconnecting or reconnecting and then performing a "sync" by entering the last focus position.  the FocusBoss II still has the last focus position store in it after powering every thing back up to start imaging  so INDI should be able to just pull the focus position from the FocusBoss II.

Really enjoying using KStars/INDI!  Thanks for all of your hard work. 

Scott

knro, 2018-08-08:
Absolute focuser?

sterne-jaeger, 2018-09-13:
I am using FocusBoss II as well with the PDSM focuser. For this focuser it works. You only have to do an initial sync. But it reads the old position from the focuser, so you simply need to copy the value to the sync field and press the button.

knro, 2018-10-26:
Please check if PR #741 fixes this issue for you.

knro, 2018-12-10:
Closing due to lack of feedback.

closed 2018-12-10 by knro