indilib/indi #669: Build error - need to link with libc6

gnthibault, 2018-08-02:
Dear all,

Thank you for developing such a nice piece of software, it has been incredibly useful for me during the past year.

I recently got a build error, and I am not sure wether it comes from my side or not:

CMakeFiles/indi_hid_test.dir/libs/indibase/hid_libusb.c.o: In function `get_usb_string':
/home/gnthibault/projects/indi/libindi/libs/indibase/hid_libusb.c:366: undefined reference to `libiconv_open'
/home/gnthibault/projects/indi/libindi/libs/indibase/hid_libusb.c:379: undefined reference to `libiconv'
/home/gnthibault/projects/indi/libindi/libs/indibase/hid_libusb.c:395: undefined reference to `libiconv_close'

It looks like indibase should be linked with a shared object that contains the symbols libiconv_open, libiconv, libiconv_close.

On my machine, it looks like I cannot find those symbols in any libiconv shared object.

It looks like, on ubuntu, content of libiconv is now part of the libc: https://askubuntu.com/questions/898826/why-is-there-no-libiconv-package-in-ubuntu

Thank you in advance for your help

knro, 2018-08-02:
Which Ubuntu are you on?

gnthibault, 2018-08-02:
Ubuntu 16.04
I saw that there is a cmake_modules/FindIconv.cmake inside indi, but it does not look like it is used inside of a find_library call ?

knro, 2018-08-02:
It's only used for Cygwin. It's not required for regular Ubuntu. Are you trying to develop for INDI? If not, it might be easier just to use the INDI PPA: http://indilib.org/download/ubuntu.html

gnthibault, 2018-08-02:
I am not actively developing something inside indi right now, but I do some experiment from time to time with the source code. I am not in a hurry to get this problem sorted out.

gnthibault, 2018-08-07:
Thank you very much for accepting the PR !
I can now close the issue.

gnthibault, 2021-01-30:
I have no idea why I am the only one to experience this issue

```
make
[  0%] Linking CXX executable indi_hid_test
CMakeFiles/indi_hid_test.dir/libs/indibase/hid_libusb.c.o: In function `get_usb_string':
/home/gnthibault/projects/indi/libs/indibase/hid_libusb.c:344: undefined reference to `libiconv_open'
/home/gnthibault/projects/indi/libs/indibase/hid_libusb.c:357: undefined reference to `libiconv'
/home/gnthibault/projects/indi/libs/indibase/hid_libusb.c:373: undefined reference to `libiconv_close'
collect2: error: ld returned 1 exit status
CMakeFiles/indi_hid_test.dir/build.make:99: recipe for target 'indi_hid_test' failed
make[2]: *** [indi_hid_test] Error 1
CMakeFiles/Makefile2:72: recipe for target 'CMakeFiles/indi_hid_test.dir/all' failed
make[1]: *** [CMakeFiles/indi_hid_test.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

```

iconv stuff is inside libc.so.6 on ubuntu:

> nm -D /lib/x86_64-linux-gnu/libc.so.6 | grep iconv
> 00000000000223e0 T iconv
> 00000000000225a0 T iconv_close
> 0000000000021fa0 T iconv_open

gnthibault, 2021-01-30:
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 832716eab..2c8deba5f 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -474,6 +474,7 @@ install(TARGETS indidriver ARCHIVE DESTINATION ${CMAKE_INSTALL_LIBDIR})
 else()
 ## Static indidriver Library
 find_package(FFTW3 REQUIRED)
+find_package(Iconv REQUIRED)
 add_library(indidriverstatic STATIC ${indidriver_C_SRC} ${indidriver_CXX_SRC} ${libstream_C_SRC} ${libstream_CXX_SRC} ${libwebcam_C_SRC} ${libwebcam_CXX_SRC} ${hidapi_SRCS} ${libdsp_C_SRC} ${fpack_C_SRC})
 set_target_properties(indidriverstatic PROPERTIES COMPILE_FLAGS "-fPIC")
 target_compile_definitions(indidriverstatic PRIVATE "-DHAVE_LIBNOVA")

```

This fixes the problem for me

TallFurryMan, 2021-01-30:
I find it strange that the iconv_open call is converted into a libiconv_open call. Seems there is a macro doing things in the background?

gnthibault, 2021-01-30:
Good point, I don't know which magic is used to glue

> nm -D ./libindidriver.so | grep iconv
>                  U libiconv
>                  U libiconv_close
>                  U libiconv_open
> 
> 

and

> nm -D /lib/x86_64-linux-gnu/libc.so.6 | grep iconv
> 00000000000223e0 T iconv
> 00000000000225a0 T iconv_close
> 0000000000021fa0 T iconv_open

But it looks like it is working

closed 2021-03-12 by knro