indilib/indi #672: Bad variable definition in INDI driver (int instead of longint) prevents focuser connexion.

jcj2249, 2018-08-03:
Intermittently I had problem connecting my Microtouch focuser to KStars : the resource was busy.
I posted the issue "Microtouch focuser refuses to connect : Resource busy" in the thread http://indilib.org/forum/focusers-filter-wheels/3692-microtouch-focuser-refuses-to-connect-resource-busy.html.
Eric (TallFurryMan) found the reason : when the focuser step counter flew over 32767 ticks, the driver didn't respond correctly anymore.
I'm using KStars 2.9.7 on macOS.
Would it be possible to correct this problem in the next release ?

knro, 2018-08-03:
THanks for the report. Looks like the driver was transformed from this: https://github.com/AzGilrock/Starizona/blob/master/MicroTouch/UsbPort.vb

However, get position is 16bit unsigned, up to 65536 ticks while the driver limits it to half of this range. Will work on a fix soon.

closed 2018-08-03 by knro

jcj2249, 2018-08-03:
Thanks Jasem !

Have a nice evening.
__________
Jean-Claude

commit d6b040f4 (2018-08-03): Fixes #672 as the driver was limited to int16_t when reading position where it should have been uint16_t