indilib/indi #674: Remove unnecessary null pointer checks

elfring, 2018-08-05:
[An extra null pointer check is not needed in functions](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for null before delete p?") like the following.
- [Dome](https://github.com/indilib/indi/blob/783f00e8dd26180eca3499783f4823d2c493ea24/libindi/libs/indibase/indidome.cpp#L67 "Destructor for Dome")
- [DSICCD::Disconnect](https://github.com/indilib/indi/blob/bab3a9fb32d344a1441847845d4dddae60123b85/3rdparty/indi-dsi/dsi_ccd.cpp#L153 "DSICCD::Disconnect function")
- [EQMod](https://github.com/indilib/indi/blob/7dabeab3a2859e20f38d7afd2f175ef160e8c8c2/3rdparty/indi-eqmod/eqmod.cpp#L210 "Destructor for EQMod")
- [FLICCD](https://github.com/indilib/indi/blob/a4539340bde871fff3565edfcd37cb5a2cc1df47/3rdparty/indi-fli/fli_ccd.cpp#L89 "Destructor for FLICCD")
- [MGenAutoguider](https://github.com/indilib/indi/blob/73a28bebbb6a6d41e7048ede869612e05a1ce49a/3rdparty/indi-mgen/mgenautoguider.cpp#L207 "Destructor for MGenAutoguider")

knro, 2018-08-05:
Can you please submit a PR?

closed 2018-08-08 by knro

elfring, 2018-08-08:
Thanks for [your small source code improvement](https://github.com/indilib/indi/commit/97c18de69824f006140f37b4713cc9f4950a9580 "Remove unnecessary null pointer checks.").

-----

How do you think about to apply a corresponding update suggestion which could be generated by [the software “Coccinelle”](http://coccinelle.lip6.fr/ "Source code transformation with Coccinelle")?

```
@Remove_unnecessary_pointer_checks@
expression x;
@@
-if (\(x != 0 \| x != NULL\))
    delXMLEle(x);
```

commit 97c18de6 (2018-08-08): Remove unnecessary null pointer checks. Closes #674