indilib/indi #676: Document tolerance for null pointers in specific function parameters

elfring, 2018-08-05:
[The function “delXMLEle”](https://github.com/indilib/indi/blob/666a5f06644d11a26aaeefce0c957d5a6c20df16/libindi/libs/lilxml.c#L163 "Implementation of delXMLEle()") was implemented in the way that it checks if a null pointer was passed as an input parameter. It returns then immediately.

Can such a detail become an official property in the affected application programming interface?
How would you like to represent information about existing input parameter validation in [the corresponding API documentation](https://github.com/indilib/indi/blob/666a5f06644d11a26aaeefce0c957d5a6c20df16/libindi/libs/lilxml.h#L94 "Update candidate")?

knro, 2018-08-08:
What's your suggestion exactly regarding this? Can you submit a PR to improve the API docs?

elfring, 2018-08-08:
* I am looking for an acknowledgement from the main developers of this software for the mentioned technical detail before.
* Other designers expressed similar details in different ways for API documentations already. Which wording will get the desired consensus?

knro, 2018-08-08:
Sorry I'm still not following, in the function documentation you want us to say that it's safe to pass nullptr?

elfring, 2018-08-08:
Yes. (The function implementation contains the comment “benign if NULL” before the relevant condition check.)

elfring, 2018-08-08:
* Have you got any wording preferences here?
* Will any tags become useful for such a function parameter property?

knro, 2018-08-08:
The actual documentation for this function is in [lilxml.h](https://github.com/indilib/indi/blob/666a5f06644d11a26aaeefce0c957d5a6c20df16/libindi/libs/lilxml.h#L94) which not correct now since it says it returns a value when it is declared as void.

closed 2018-08-08 by knro

elfring, 2018-08-08:
Why have you closed this issue?

elfring, 2018-08-08:
Will it make sense to add null pointer tolerance also for the parameter “lp” of [the function “delLilXML”](https://github.com/indilib/indi/blob/eb29df827eb9ef00bace77229d5a406c6359bc7d/libindi/libs/lilxml.h#L89 "Checking function properties")?

commit db52aded (2018-08-08): Fix API documentation. Closes #676