indilib/indi #678: NexStar driver doesn't return Side of pier

basilr, 2018-08-08:
The Side of pier status is used by PHD Guiding to determine how to send guiding pulses. Right now, after a meridian flip PHD must be re-calibrated each time (really slowing down the imaging run). 

In my case (a C14 on CGE Pro mount), if I don't do a re-calibration after a meridian flip the RA corrections are reversed and the guide star instantly starts going the wrong way in RA causing guiding to fail.

knro, 2018-08-08:
It's not reported by the protocol AFAIK. You have an idea on how to implement this for the driver?

basilr, 2018-08-08:
I am not a programmer, but I hope the helps.

I know PHD works with ASCOM, so I looked at the ASCOM driver docs for the Celestron mounts. It seems Celestron did implement SideOfPier in the newest versions (after the NexStar communication protocol doc was written - the doc referenced in the INDILib Devices webpage).

Here's an extract from the ASCOM Celestron driver doc (I'm attaching the full PDF):

SideOfPier and DestinationSideOfPier are only applicable to the GEM scopes. 

"For HC versions 4.15 or more SideOfPier and DestinationSideOfPier are read from the 
mount. For the CGEM and CGE Pro SideOfPier can also be set - if the direction the scope is 
pointed to is accessible from the side of the pier specified then the meridian preference 
commands are used to move the scope to that side of the pier. "

So _I Think_ if someone is using a Hand Controller (HC) version 4.15 (which has been out for a while) then SideOfPier is reported.

If you'd like to write a Beta driver I'd be happy to test with my CGE Pro mount.

Hope this helps!
Thanks
Basil
[ASCOM.Celestron.Telescope.pdf](https://github.com/indilib/indi/files/2270405/ASCOM.Celestron.Telescope.pdf)

basilr, 2018-08-08:
I'm investigating... 

not being a programmer, but...

Are you looking for the specific commands returned when querying the mount? e.g. sideofpier=1 when east pointing west, sideofpier=-1 when west pointing east, etc...

I might forward this on to the developer of PHD, Andy Galasso, for his thoughts.

agalasso, 2018-08-08:
I reached out to Chris Rowland, the developer of the Celestron Unified ASCOM Driver to see if he can provide the needed protocol information.

agalasso, 2018-08-08:
I came across this:
http://www.paquettefamily.ca/nexstar/NexStar_AUX_Commands_10.pdf
How about using the pass-through command "MC_GET_POSITION" to get the encoder position for the dec axis then use that to deduce the pointing state in the same way as EQMod::EncodersToRADec()

knro, 2018-08-09:
Thanks @agalasso. I think I'll wait for @juanmb  since he made the recent changes in the driver and probably knows how to include this much better than I do

juanmb, 2018-08-10:
That's very interesting. This command is not documented in the [protocol manual I followed](http://www.nexstarsite.com/download/manuals/NexStarCommunicationProtocolV1.2.zip).
It is a very important feature that I missed in my mount (a GC5-GT).
I'm quite busy right now, but I will try to work on it next week.

basilr, 2018-08-10:
Thanks to everyone for the collaboration on this!
If there is anything I can do to help please let me know. I'd be happy to test any beta versions with my C14 and CGE Pro mount (I have a version 4.15 HC).

knro, 2018-08-28:
@juanmb I know you're probably busy but any insights on this?

juanmb, 2018-08-29:
Yes, I've been quite busy, as always :(
Also, I had to fix some problems in my mount controller before trying to implement this feature. The main problem is I don't have a working hand controller, and my mount is an Advanced GT, so I implemented the PierSide command in my Arduino-based controller, trying to mimic the response of a CGEM.
I've been looking over the code of INDI::Telescope, and It should be easy to implement. Tomorrow I will work on it.

juanmb, 2018-08-31:
I've uploaded a new commit to my repository under the branch [feature/pierside](https://github.com/juanmb/indi/tree/feature/pierside). @basilr could you try to test it with your mount? You'll have to compile the driver from source code.

I have some doubts about which mount models respond to the *p* command. The models are defined [here](https://github.com/juanmb/indi/blob/e1ebbccdb489bc6311f14f393e75160e052e0fed/libindi/drivers/telescope/celestrondriver.cpp#L325). Currently I enabled this feature for *CGE/CGEM* mounts with hand control version >= 4.15. The condition is tested [here](https://github.com/juanmb/indi/blob/b64d6fca37eb5de0599c60a0935db9d4ba8e9b54/libindi/drivers/telescope/celestrongps.cpp#L246).

basilr, 2018-09-01:
@juanmb thanks very much for doing this. I'm afraid I'll need a little hand holding on "compile the driver from source" part. 
I know Linux, comfortable with a terminal window, etc.. I can run shell scripts, etc. and have compiled stuff before - but I've needed step by step instructions. Also, I'm a noob using Github.

I've downloaded the fork with the new commits, read the Readme and tried to follow what I think I need to do.., but then I get lost. I know this should be pretty easy, but...

Any help is greatly appreciated!

juanmb, 2018-09-01:
OK. First of all, you have to install some prerequisites. If you are using Ubuntu, Debian or similar, you should be able to install them all by executing this command:
```
sudo apt-get install libnova-dev libcfitsio-dev libusb-1.0-0-dev zlib1g-dev libgsl-dev build-essential cmake git libjpeg-dev libcurl4-gnutls-dev
```
Then you can clone my git repository and "checkout" the branch where I am working:
```
git clone https://github.com/juanmb/indi.git
cd indi
git checkout feature/pierside
```
Then you can compile the indi server with the standard drivers:
```
cmake -DCMAKE_BUILD_TYPE=Debug .
make
```
I suggest you not to execute ```make install``` as indicated in the  [README](https://github.com/indilib/indi/blob/master/README.md), because it will interfere with your previously installed indi packages.
Instead, you can run your new `indiserver` manually by running
```
./indiserver -v ./indi_celestron_gps
```
From the Device Manager in KStars you can now connect to the indiserver in `localhost`, port 7624. Make sure KStars doesn't start its own indiserver, because it will try to run the old one.

If everything works, you should see a new "Pier side" control under "Celestron GPS" -> "Main control" tab in INDI Control Panel.
Hope this helps!

basilr, 2018-09-02:
Your instructions were great. Compiling/building worked!

At the scope: I started the indiserver with the Celestron driver, connected with indi client (PHD). Saw "Pier Side" in Main Control. Showed East (pointing west) which was correct.Great!

However slewed to other side of pier. Still said East pointing West, which was wrong. Tried disconnecting and reconnecting to the indi server - no change. Tried stopping indiserver, then starting again with scope on West side pointing east. indi still says East pointing west. Tried rebooting my Linux (actually a RaspberryPi running Raspian) still shows East pointing west. Tried running the indiserver with sudo...

Nothing funny showing up in the indiserver log window. Is there a log file or debug file I can send you??

Progress!!
Thanks

juanmb, 2018-09-02:
Maybe the mount is not responding to the "p" command, or the response is not what I expected.
You can watch [this video](http://indilib.org/support/logs-submission.html) in which Jasem shows how to record a log from an INDI driver. Please mark all the options (like "Driver Debug" and "Scope verbose").
Maybe the log could allow me to find out what's going on.

basilr, 2018-09-03:
So I think I was wrong, it looks like things are working.

My previous test was based on me using the hand controller alone to move the mount as I watched the indi control panel. No change in the control panel. This should be expected?

Tonight I gathered a debug log (attached) and used the Ekos mount control to move the mount. As I slewed around I noticed the indi control panel change from East pointing west to West pointing east! I believe this is also recorded in the log file. So I think it's working as it should...

What I need to do now is get my other drivers for my CCD and guider (SBIG, ZWO) running with the new  Celestron driver and try using PHD to guide on both sides of the pier. I hope to try this tomorrow night.

Looks promising. 

The log shows 2 sessions, one with the normal driver (NexStar), the second with the test GPS driver and pier side.

Many thanks.
[log_20-49-54.txt](https://github.com/indilib/indi/files/2344135/log_20-49-54.txt)

juanmb, 2018-09-03:
Looking at the log, I can see the mount is returning the expected response. It seems the driver is working properly.
I don't know why the driver does not update the Pier Side field when you move the telescope using the hand control. I didn't expected that!

basilr, 2018-09-04:
Last nights test went pretty well. PHD handled the meridian flips.

I want to do some more testing (tonight?). It seems there was **clearly** a need to tell PHD to reverse Dec output after a meridian flip
[image]

However some of my first tests didn't seem to need that option... I want to clear this up. 

Also, it seemed I needed to do a re-calibration in PHD when I turned on the Reverse Dec option. I also want to clear this up.

All in all, I was able to use the new Celestron driver sucessfully to do all my imaging last night (asteroid photometry, if anyone is interested...). But had some issues (probably user error) I want to try and resolve.

Weather looks good for a run tonight...

Thanks!

agalasso, 2018-09-04:
> However some of my first tests didn't seem to need that option... I want to clear this up.

> Also, it seemed I needed to do a re-calibration in PHD when I turned on the Reverse Dec option. I also want to clear this up.

There's a quick procedure in the PHD2 help file to determine the appropriate value for the setting. https://openphdguiding.org/manual/?section=Advanced_settings.htm#Guiding_Tab (scroll down to 'Reverse Dec output after meridian flip' )

basilr, 2018-09-05:
Last night's session went great! Everything worked as expected. Thanks @agalasso for the info on the Reverse Dec setting. Guiding was SO much easier with meridian flips. This is a big deal. Thank You to everyone! I'd be happy to do whatever needed before going live with the new driver.

One final thought as I was going through my normal setup routine:

I would be very helpful to be able to set PEC playbeck with INDI. This setting is not remembered when hibernating or powering off the mount. Turning on PEC playback is the only thing I have to pick up the hand controller and do. The ability to set this with INDI would make all the functionality needed available remotely (for me and probably most others) . I'm pretty sure PEC commands are available according the the document @agalasso referenced earlier
http://www.paquettefamily.ca/nexstar/NexStar_AUX_Commands_10.pdf

Anyway, **Many thanks** again!

juanmb, 2018-09-05:
I'm glad everything is working fine!
I have to review the code and then I'll make a pull request to the main INDI repository, so Jasem could include it in the next release.

juanmb, 2018-09-05:
As for the PEC issue, frankly, I don't know how to do that :disappointed: 
My mount (Advanced GT) does not support PEC correction, so the trial and error process would be very tedious. Maybe another developer with a CGE/CGEM mount could give it a try!

closed 2018-09-06 by basilr

Chris-Rowland, 2018-09-22:
I will try to review this once I've got a development environment working.  Getting the pointing state (aka Pier side) is pretty easy using the 'p' command but will only work with GEMs.  It's easy to work this out from the mount model.  The list of models needs updating.  Adding PEC control also shouldn't be a problem, I can test with my AVX mount.

I see this is closed, what is involved in getting this into the main branch?

Chris

knro, 2018-09-23:
The next step is to send a PR along with the update mount list codes and a way to separate equatorial mounts from others so that the pier property is conditionally defined after identifying the model in use.

Chris-Rowland, 2018-09-23:
Would it help if I added a better check for the mounts that can implement pier side?  I would update the model list and add a bool isGEM property to FirmwareInfo.  I'll also review how I handled the Southern Hemisphere in the ASCOM driver.  From what I recall I may have needed to do some version dependent shenanigans.

Not used Git before so it may take a while for me to get this done.

Chris

knro, 2018-09-23:
That would be great! For GIT, fork the INDI repo and make changes to your local copy and commit.. once you think it is ready, you can use the Github interface to create a "Pull Request" to merge your changes back into the repository.

Chris-Rowland, 2018-10-11:
I've done some fixes to reading of PierSide and they have been pushed to the main thread and a bug has been fixed.  I don't think there is more I can do on this issue.

I've done nothing about PEC but have been looking at other things, in particular allowing the INDI Tracking Mode to control the tracking rates - Sidereal, Solar and Lunar - instead of the Celestron tracking mode which is really to do with setting the type of mount.

On a admin note this is all on a closed issue, should it be opened?  And should the other things - Tracking rates and PEC have their own issues?

Chris