indilib/indi #680: reserved identifier violation

elfring, 2018-08-08:
I would like to point out that an identifier like “[`_LilXML`](https://github.com/indilib/indi/blob/eb29df827eb9ef00bace77229d5a406c6359bc7d/libindi/libs/lilxml.h#L72 "Update candidate")” [does eventually not fit](https://www.securecoding.cert.org/confluence/display/c/DCL37-C.+Do+not+declare+or+define+a+reserved+identifier "Do not use identifiers which are reserved for the compiler implementation.") to the expected naming convention of the C language standard.
Would you like to adjust your selection for unique names?

knro, 2018-08-08:
What? who else uses _LilXML in the standard C language?

elfring, 2018-08-08:
How do you think about to avoid that this software [depends on undefined behaviour](https://www.securecoding.cert.org/confluence/display/c/CC.+Undefined+Behavior#CC.UndefinedBehavior-ub_106 "Avoiding undefined behaviour?")?

closed 2018-08-08 by knro

knro, 2018-08-08:
Ok thanks. I didn't know that _ followed by capital letter makes a reserved identified. Btw, is this audit part of some ongoing project?

elfring, 2018-08-08:
Thanks for [your small source code improvement](https://github.com/indilib/indi/commit/fa60f910229494c5eb8e5552ad98eb858c1992f6 "Renaming of three data structure identifiers").

commit fa60f910 (2018-08-08): Do not use reversed identifier according to C standard. Closes #680