indilib/indi #681: Detector crashes when trying to run focusing

jochym, 2018-08-09:
If I run ekos with simulators and try to focus the moment the routine detecting stars run it crashes and takes whole indi server with it. You cannot restore functionality without restarting indi. This happens regardless if you run auto-focus or manually select star to focus. The moment you click the process crashes. If you need more data I will run it with fuller debugging and check everything in detail. This is just to mark the problem for future reference.

knro, 2018-08-09:
Thanks for the feedback. KStars/Ekos bugs need to be filed over https://bugs.kde.org

If you attach the backtrace it would help greatly.

closed 2018-08-09 by knro

jochym, 2018-08-09:
I will. You are saying this is likely bug in ekos/kstars? Ok. I'll investigate further and submit it in appropriate place.

knro, 2018-08-09:
Yes. Ekos is one client of INDI out of many, so if it crashes, the issue is reported to KDE bug tracker.