indilib/indi #683: 3rdparty build is failing on Raspbian - QSI camera is then not being recognised

phomer60, 2018-08-19:
This started 1-2 weeks ago.
I have cleared the build directory a number of times and istalled all the required packages that I am aware of of. As I install for all devices I run through the installation twice. Below is partial output from each step.
**cmake**
-- Could not find up to date FFMPEG, could be due to libavcodec, libavdevice, libavformat, libavutil, or libswscale version
-- Since an up to date FFMPEG was not found, INDI Webcam Driver will not be built
...
-- ####################################################################################################################################
-- Not all libraries found, you must build and install all libraries first:
-- libqhy was not found and will now be built. Please install libqhy first before running cmake again to install indi-qhy.
-- ####################################################################################################################################
**make**
[ 82%] Building CXX object indi-rtlsdr/CMakeFiles/indi_rtlsdr_detector.dir/indi_rtlsdr_detector.cpp.o
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp: In member function ‘void RTLSDR::grabData()’:
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp:383:5: error: ‘dspau_stream_p’ was not declared in this scope
     dspau_stream_p stream = dspau_stream_new();
     ^~~~~~~~~~~~~~
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp:384:26: error: ‘stream’ was not declared in this scope
     dspau_stream_add_dim(stream, len);
                          ^~~~~~
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp:384:37: error: ‘dspau_stream_add_dim’ was not declared in this scope
     dspau_stream_add_dim(stream, len);
                                     ^
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp:392:43: error: expected primary-expression before ‘unsigned’
     dspau_convert_from(data8, stream->in, unsigned char, len);
                                           ^~~~~~~~
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp:392:61: error: ‘dspau_convert_from’ was not declared in this scope
     dspau_convert_from(data8, stream->in, unsigned char, len);
                                                             ^
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp:393:48: error: ‘dspau_filter_squarelaw’ was not declared in this scope
     stream->out = dspau_filter_squarelaw(stream);
                                                ^
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp:394:42: error: expected primary-expression before ‘unsigned’
     dspau_convert_to(stream->out, data8, unsigned char, len);
                                          ^~~~~~~~
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp:394:60: error: ‘dspau_convert_to’ was not declared in this scope
     dspau_convert_to(stream->out, data8, unsigned char, len);
                                                            ^
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp:398:43: error: ‘magnitude_root’ was not declared in this scope
     spectrum = dspau_fft_spectrum(stream, magnitude_root, 256);
                                           ^~~~~~~~~~~~~~
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp:398:62: error: ‘dspau_fft_spectrum’ was not declared in this scope
     spectrum = dspau_fft_spectrum(stream, magnitude_root, 256);
                                                              ^
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp:404:29: error: ‘dspau_stream_free’ was not declared in this scope
     dspau_stream_free(stream);
                             ^
indi-rtlsdr/CMakeFiles/indi_rtlsdr_detector.dir/build.make:62: recipe for target 'indi-rtlsdr/CMakeFiles/indi_rtlsdr_detector.dir/indi_rtlsdr_detector.cpp.o' failed
make[2]: *** [indi-rtlsdr/CMakeFiles/indi_rtlsdr_detector.dir/indi_rtlsdr_detector.cpp.o] Error 1
CMakeFiles/Makefile2:1991: recipe for target 'indi-rtlsdr/CMakeFiles/indi_rtlsdr_detector.dir/all' failed
make[1]: *** [indi-rtlsdr/CMakeFiles/indi_rtlsdr_detector.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

phomer60, 2018-08-19:
Attached is the log file
[log_16-05-17.txt](https://github.com/indilib/indi/files/2300118/log_16-05-17.txt)

jochym, 2018-08-19:
It looks like you are missing some libraries and did not observe correct build procedure. 
You need to:
1. Build and _install_ core libindi
2. Build and _install_ 3rd party libraries
3. Build and install 3rd party drivers

Until you have 3rd party libraries build and _installed_ you will not be able to compile some 3rd party drivers. This is a deficiency of our build system but there is nobody at the moment wiling to rework this system to a better one. And since we try to provide binary distros for the users it seems to be of low importance. Why you are not using binary packages for the Pi? What we should add to the pi distro to avoid compilation by the users?

phomer60, 2018-08-19:
I have cleared the build directories for libindi and 3rdParty and started over again but I am still experiencing the same issue. I think the problem is that it is not completing the 3rdparty build because oif the error above. I have tried this only a number of microSD cards and one a Raspberry Pi 2 and 3+ and still experiencing problems.

knro, 2018-08-19:
You probably have older libindi somewhere under /usr/local

Search there and delete any INDI related files/library, then try again. No need to build the whole 3rd party folder, just go to indi-qsi and run cmake there

phomer60, 2018-08-19:
Jasem,

I can't find anything under /usr/local that is INDI related.

knro, 2018-08-19:
Do you have SSH or TeamViewer where I can check it out?

phomer60, 2018-08-19:
Jasem,

I may be wrong but this all started happening when the make started failing to complete as seen below.
[ 82%] Built target indi_gpsnmea
[ 82%] Built target indi_mgenautoguider
[ 82%] Building CXX object indi-rtlsdr/CMakeFiles/indi_rtlsdr_detector.dir/indi_rtlsdr_detector.cpp.o
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp: In member function ‘void RTLSDR::grabData()’:
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp:383:5: error: ‘dspau_stream_p’ was not declared in this scope
     dspau_stream_p stream = dspau_stream_new();
     ^~~~~~~~~~~~~~
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp:384:26: error: ‘stream’ was not declared in this scope
     dspau_stream_add_dim(stream, len);
                          ^~~~~~
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp:384:37: error: ‘dspau_stream_add_dim’ was not declared in this scope
     dspau_stream_add_dim(stream, len);
                                     ^
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp:392:43: error: expected primary-expression before ‘unsigned’
     dspau_convert_from(data8, stream->in, unsigned char, len);
                                           ^~~~~~~~
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp:392:61: error: ‘dspau_convert_from’ was not declared in this scope
     dspau_convert_from(data8, stream->in, unsigned char, len);
                                                             ^
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp:393:48: error: ‘dspau_filter_squarelaw’ was not declared in this scope
     stream->out = dspau_filter_squarelaw(stream);
                                                ^
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp:394:42: error: expected primary-expression before ‘unsigned’
     dspau_convert_to(stream->out, data8, unsigned char, len);
                                          ^~~~~~~~
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp:394:60: error: ‘dspau_convert_to’ was not declared in this scope
     dspau_convert_to(stream->out, data8, unsigned char, len);
                                                            ^
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp:398:43: error: ‘magnitude_root’ was not declared in this scope
     spectrum = dspau_fft_spectrum(stream, magnitude_root, 256);
                                           ^~~~~~~~~~~~~~
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp:398:62: error: ‘dspau_fft_spectrum’ was not declared in this scope
     spectrum = dspau_fft_spectrum(stream, magnitude_root, 256);
                                                              ^
/home/pi/indi/3rdparty/indi-rtlsdr/indi_rtlsdr_detector.cpp:404:29: error: ‘dspau_stream_free’ was not declared in this scope
     dspau_stream_free(stream);
                             ^
indi-rtlsdr/CMakeFiles/indi_rtlsdr_detector.dir/build.make:62: recipe for target 'indi-rtlsdr/CMakeFiles/indi_rtlsdr_detector.dir/indi_rtlsdr_detector.cpp.o' failed
make[2]: *** [indi-rtlsdr/CMakeFiles/indi_rtlsdr_detector.dir/indi_rtlsdr_detector.cpp.o] Error 1
CMakeFiles/Makefile2:1991: recipe for target 'indi-rtlsdr/CMakeFiles/indi_rtlsdr_detector.dir/all' failed
make[1]: *** [indi-rtlsdr/CMakeFiles/indi_rtlsdr_detector.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

koichifun, 2018-08-20:
If you don't need indi_rtlsdr, simply turn off the option to build it in CMakeLists.txt under indi/3rdparty/.
If you use indi_rtlsdr, do the followings:
The error is due to the out-of-date libdspau, so delete libdspau.* under /usr/lib/arm-linux-gnueabihf, then cmake for all 3rdparty drivers.
You'll find a message "libdspau was not found and will now be built. Please install libdspau first before running cmake again to install indi-rtlsdr".
Now "libdspau" directory is made under the build directory for the 3rdparty drivers. 
Go into the libdspau directory and do "make" and "sudo make install". Then go back to the 3rdparty directory, you'll sucessfully "make" and "sudo make install".

koichifun, 2018-08-23:
phomer60, 
Are libdspau.* installed in any of the following direcotories?
/usr/lib/arm-linux-gnueabihf
/usr/lib/
/usr/local/lib, or anywhere else. The directory depends on how you installed libdspau.
If you didn't find the message "libdspau was not found and will now be built. ", they should be somewhere, and they would be out-of-date.

phomer60, 2018-08-25:
I have rebuilt everything from scratch and instead of building all 3rdparty I have tried just building indi-qsi. By everything I mean from Raspbian up.
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ../../3rdparty/indi-qsi
-- Found CFITSIO: /usr/lib/arm-linux-gnueabihf/libcfitsio.so
CMake Error at /home/pi/indi/cmake_modules/FindQSI.cmake:44 (message):
  QSI not found.  Please install libqsi http://www.indilib.org
Call Stack (most recent call first):
  CMakeLists.txt:13 (find_package)
----
Also '/usr/lib/arm-linux-gnueabihf' exists after the complete install.

tstibor, 2018-08-25:
Have you followed the build order as suggested by @jochym:

1. Build and install core libindi
2. Build and install 3rd party libraries
3. Build and install 3rd party drivers

This is crucial otherwise otherwise the compilation will fail

phomer60, 2018-08-26:
I have built and installed  libindi but I decided to did it again in case something failed that I did not notice. I am still getting the same errors.
I am trying to do this without building all 3rdparty and the first I am trying to do is 'indi-qsi'. I believe I am following the instructions correctly but feel I must be missing something.

knro, 2018-08-26:
It says it's missing libqsi, so build it like indi-qsi, it's in 3rd party repo.

phomer60, 2018-08-26:
I still have a lot of work to get things set up but I now have the QSI camera working. I now have a clearer idea about how this works so the effort has been worthwhile.
I know there was something I had to install to get the icons working but can't recall what I did.

knro, 2018-08-28:
This is another issue, I think there is thread on INDI forum regarding that. For now this issue is closed.

closed 2018-08-28 by knro