indilib/indi #690: CEM120 @ FW 180620;    Mount will not respond to this version request ":V#" and therefore will never pass checkconnection.

DZeek86, 2018-08-26:
https://github.com/indilib/indi/blob/9845ed647cf01685c60c2a075ac909219604b712/libindi/drivers/telescope/ioptronv3driver.cpp#L126

knro, 2018-08-26:
What firmware supports this command?

DZeek86, 2018-08-26:
I'm not sure off hand?  I see ":V#" in the V2.0 command language doc, but when sent to my mount it will not respond to  ":V#"  I just received the mound and have August 8th 2018 firmware release.   Not sure if previous FW version works.  I've rewritten to use :MountInfo# on my local computer to get by for now..

DZeek86, 2018-08-26:
I checked agains a captured start up with iOptron ASCOM driver and looked like was using :MountInfo# .. maybe the version check ":V#" is being dropped from command list..not sure.

knro, 2018-08-26:
What's the response for :MountInfo ?

DZeek86, 2018-08-26:
It looks like its the model identifier:  "0120" corresponds to the Driver::models constant table.  ":MountInfo#" is in the V2.0 command language doc.

DZeek86, 2018-08-26:
Here's the fix up I used to get by to use the CEM120 for now:  I have it fixed to do a check against "0120" for return of of check.. maybe could check against the find models list instead.  Also, had to change the send command to look for a 4 character return as it doesn't return "#" and had to fix up the send command to drop through on a 4 character count check at the bottom of that routine.

    for (int i = 0; i < 2; i++)
    {
//        if (sendCommand(":V#", -1, res, 3) == false)
        if (sendCommand(":MountInfo#", 4, res, 3) == false)
        {
            usleep(50000);
            continue;
        }

//        return (!strcmp(res, "V1.00"));
        return (!strcmp(res, "0120"));
    }

    return false;
}

commit bd016a1e (2018-08-26): Use :MountInfo# to check connection for iOptronV3

Fixes #690
closed 2018-08-26 by knro