indilib/indi #693: FitsViewer does not respect Fits standard

djibb, 2018-08-28:
When you visualize a fits file with personalized min/max value, FITS keywords DATAMIN and DATAMAX are changed in file concerned.
But DATAMIN and DATA MAX are fits standards keywords 
https://fits.gsfc.nasa.gov/fits_standard.html

KEYWORD: DATAMAX
REFERENCE: FITS Standard
STATUS: reserved
HDU: image
VALUE: real
COMMENT: maximum data value
DEFINITION: The value field shall always contain a floating point
number, regardless of the value of BITPIX. This number shall give the
maximum valid physical value represented by the array, exclusive of any
special values.

So DATAMIN and DATAMAX can't be used to store theses values as it correspond to camera numeric depth. (for example, for a 16 nits camera, value are DATAMIN = 0 and DATAMAX = 65535 )

For example, Audela store it in MIPS-MAX mans MIPS-MIN keywords. (not in standards...)
May be Kstars can do the same... or store it in other non-standards keywords.

TallFurryMan, 2018-08-28:
I agree, DATAMIN/DATAMAX do not relate to the actual minimum/maximum values of FITS data, but indicate what the dynamic range of the physical equivalent of the FITS data is. So they can't be changed at all.

Not that I trust particularly that plug-in, but here's a side reference on the GIMP plug-in reading FITS files: https://gitlab.gnome.org/GNOME/gimp/blob/master/plug-ins/file-fits/. That plug-in didn't change much since 1999...

FITS header is read in https://gitlab.gnome.org/GNOME/gimp/blob/master/plug-ins/file-fits/fits-io.c. There are two modes, stated "automatic" and "by DATAMIN/DATAMAX":
- With "automatic", FITS data is applied a reference change from the minimum/maximum value of FITS data to the [0,DATAMAX] interval. This is just weird as an automatic mode.
- With "by DATAMIN/DATAMAX", FITS data is applied a reference change from origin BZERO and scale BSCALE. The minimum/maximum value of FITS data is considered to be the inverse reference change of DATAMIN/DATAMAX. This is proper conversion to physical data values.

TallFurryMan, 2018-08-28:
In any case, this is a bug related to the KStars FITS Viewer that should be opened at bugs.kde.org, not here.
This ticket will probably be closed soon.

knro, 2018-08-28:
Thanks for the report.. as @TallFurryMan indicated above me, this needs to go to bugs.kde.org

closed 2018-08-28 by knro