indilib/indi #711: FITS header DATE-OBS has wrong format

sterne-jaeger, 2018-09-13:
Frames contain the observation time in the FITS header as time stamp with a trailing 'Z' for Zulu time. According to the FITS specification, the time stamp must be pure without trailing time zone information.

This causes trouble in the PixInsight, where the comet alignment module obtains the date in the wrong format and crashes. They should make it more robust, but it would be good if it is fixed in INDI.

glowmouse, 2018-10-02:
I'm not an indi developer,  but I tried this on my Ubuntu 18.04.1 Linux machine, with libcfitsio-dev version 3.430-2, I see the following when I use the CCD Simulator to generate a test image:

DATE-OBS= '2018-10-02T11:08:35.241' / UTC start date of observation

This seems to comply with the date standard in https://fits.gsfc.nasa.gov/standard30/fits_standard30aa.pdf, Section 4.4.2.1.

Copying the OBS-DATE that you're seeing into the ticket help.  Information about the OS and indi version you're running on might also help narrow  down the root cause of the bug.

glowmouse, 2018-10-03:
I did some digging... I'd' swear this field is set here:

const char *CCDChip::getExposureStartTime()

In this code:
    // Format it in ISO8601 format
    strftime(iso8601, sizeof(iso8601), "%Y-%m-%dT%H:%M:%S", tp);

    // Add millisecond and Z for Zulu/UTC time.
    snprintf(ts, 32, "%s.%03dZ", iso8601, static_cast<int>(startExposureTime.tv_usec / 1000.0));

But the fits files that I'm generating are missing the Z... If nobody chimes in I'll look into it more this weekend - it seems like a good way to learn the code.

knro, 2018-10-03:
Thanks for the report. I'll remove the Zulu postfix if it's causing issues.

commit 07f9203c (2018-10-03): Remove Zulu postfix. Fixes #711
closed 2018-10-03 by knro