indilib/indi #712: seletek does not work with autofocus

Quilex, 2018-09-13:
Hi. 
Sorry mi bad English.

Web traslator.

When trying to perform autofocus, the camera makes the first capture, seletek moves the focuser and the process remains stopped.
It is as if the camera is waiting for Seletek to notify him that he has finished moving the focus.
It does not produce any error that I see.

Tested on 2 computers; (Mint 19)
- original seletek + qhy9;
- seletek armadillo + Atik
 
Thanks in advance.

------------

Al intentar realizar autofoco, la camara realiza la primera captura, seletek mueve el enfocador y el proceso se queda parado.
Es como si la camara se quedara esperando a que seletek le avise que termino de mover el foco. 
No produce ningun error que yo vea.

Probado en 2 ordenadores; (Mint 19)
- seletek original + qhy9;
- seletek armadillo + Atik
 
Gracias anticipadas.

mmastria, 2018-09-30:
Same issue here.
If I move the focuser via Indi Control Panel, it reports Ok when position is reached (log window). 
If I try to autofocus, Ekos behaviour as the focuser never reaches the desired position (wait indefinitely). The focuser itself moves correctly and stop at the correct position.

knro, 2018-09-30:
Enable logging and then attach log. Does the LED light for the position property become "Green" or remains at "Yellow"?

mmastria, 2018-09-30:
> Enable logging and then attach log. Does the LED light for the position property become "Green" or remains at "Yellow"?

I can't replicate this right now (weather conditions). Maybe only the next weekend. It's a valid test if I use a "simulator CCD" with "real armadillo focuser"? I think this doesn't make sense (as a valid test).

Quilex, 2018-10-15:
I regret not having written before, I am moving from home and I have the telescope dismantled.
As soon as I can, I'll send a log,
Greetings.

mmastria, 2018-11-03:
I uploaded this log. During this session, when I tried to do auto-focus, the driver reported "Focuser reached requested position.", the light got back to green but Ekos didn't recognised and wait forever.

This log have several moves because I commanded each move "manually", take a snapshot, and moved again and again.

I think (not sure) that this first block was auto-focus that failed, no clues:
 
>DEBUG   10920.389376 sec    : Abs move to 50360
DEBUG   10920.389571 sec    : Tx [!step goto 0 50360 0#]
DEBUG   10920.406129 sec    : Rx [!step goto 0 50360 0:0#]
INFO    10920.406291 sec    : Focuser is moving to position 50360
DEBUG   10920.485697 sec    : Tx [!step getpos 0#]
DEBUG   10920.499997 sec    : Rx [!step getpos 0:50269#]
DEBUG   10920.517207 sec    : Tx [!step getpos 0#]
DEBUG   10920.532613 sec    : Rx [!step getpos 0:50272#]
DEBUG   10921.501216 sec    : Tx [!step getpos 0#]
DEBUG   10921.522406 sec    : Rx [!step getpos 0:50360#]
DEBUG   10921.532882 sec    : Tx [!step getpos 0#]
DEBUG   10921.547513 sec    : Rx [!step getpos 0:50360#]
INFO    10921.547740 sec    : Focuser reached requested position.
DEBUG   10921.547970 sec    : Tx [!read temps 1#]
DEBUG   10921.562603 sec    : Rx [!read temps 1:294#]

The other moves was commanded by me:

>DEBUG   10945.281240 sec    : Tx [!read temps 1#]
DEBUG   10945.296211 sec    : Rx [!read temps 1:294#]
INFO    10963.382501 sec    : Focuser is moving to position 50460
INFO    10964.852433 sec    : Focuser reached requested position.
INFO    10984.873556 sec    : Focuser is moving to position 50360
INFO    10986.469650 sec    : Focuser reached requested position.
INFO    10992.323293 sec    : Focuser is moving to position 50260
INFO    10993.664562 sec    : Focuser reached requested position.

When the weather permits, I'll try a better log. But yes, the green light works OK, and message that focuser reached position works too.

Mastria.

Quilex, 2018-11-04:
Hi.
I continued with the change of address, but I follow the thread.

> ////  mmastria say.
> 
> Maybe an important information:
> 
> Few months ago, the auto-focus worked perfectly "only" one time.
> In this event, I was running Ekos on the same hardware (Linux Desktop) where armadillo was plugged in (USB). In all other events I'm using in a cascading connection:
> 
> //// End say.

Are you saying that you connected the Armadillo directly to the computer and it worked?

I always connect the cable to a USB hub together with other devices, ccd, etc ... and connect it to the computer.

I can ask a friend of mine to try the direct connection.

Greetings.

mmastria, 2018-11-04:
Quilex, there's 2 scenarios here:

1 - Armadillo connected to computer where Ekos is running (with or without usb-hub)
2 - Armadillo connected to raspberryPi running indiserver, and Ekos running in other computer connecting to raspberryPi remotely (over network).

Your scenario is the #1. My scenario "today" is #2.

What I said is that auto-focus worked one time only, and, by coincidence or not, it was connected on the USB port of computer that was running Ekos (scenario #1). 

When the weather permits and I'll try several scenarios to se if I can identify at least one situation that auto-focus works. If I can do that, with logs running in all scenarios, I can post here something that really helps knro to identify the problem.

Mastria

mmastria, 2018-11-18:
Today I could take some tests. Attached some logs and pictures.
[log.zip](https://github.com/indilib/indi/files/2592531/log.zip)

I indetified what happening, but not why.

There's 4 tests in log file:
1 - Moved focuser in and out from control panel
2 -Moved focuser in and out from Ekos client (focus Tab)
3 - First autofocus try, after send "move command" stop waiting forever
4 - A successfull autofocus using a "trick" to bypass the waiting situation.

The problem occur only when Ekos send "move" command to indi. The driver moves and in driver log there's information about "reached position", but not in Ekos log on screen. In this situation, the "Absolute Position" ligth stays YELLOW. (see below).

[image]

The "trick" I used was, after each focuser move, I manually sent 1 tick move. After this, the YELLOW ligth change to GREEN and Ekos continues the autofocus procedure. Doing this after each Ekos move command to focuser and finally I got the focus. (see below)...

[image]

So the problem is, when Ekos send a "move" command, the light stays YELLOW, when I send a "move" command from Control Panel, the ligth changes to GREEN.

I hope these logs and report help to solve the problem.

Thanks,
Mastria.

mmastria, 2018-11-25:
I could test tonight, it worked perfectly (I tested only "main" focuser, but I believe that the "exp" will work too). You can close this issue.

Thanks again!
Mastria

Quilex, 2018-11-26:
Tnx Mmastria and Knro for help.