indilib/indi #713: Potential bug with auto meridian flip wrapping up cables.

focaldepth, 2018-09-24:
Potential bug with auto meridian flip wrapping up cables.

I did a forced Meridian flip goto before midnight then Auto Meridian Flip was set. Auto flip kicked in OK. But the mount took the long way round and wrapped up both RA and DEC axis.

No mount limits were set.

Versions
Ubuntu : 18.04.1, up to date.
kstars : V 6:2.9.8
indi-full : V 1.7.4
astrometry.net : V 0.73
HEQ5: old mount but new controller card fitted last year so latest version assumed.

I am very new to INDI, EKOS, KSTARS but have quite a few years experience imaging with ASCOM, EQMOD on windows. Also much experience with linux.
This was my first night using INDI etc. so beware of the NOOB factor.

I was amazed how quickly I got the scope set up, solving, guiding and imaging IC1396, all perfect. I was using VNC to monitor. Then at meridian flip time the flip started and appeared to work well but the solver started to solve various NGCxxx targets so I went outside for a look and the steppers were screaming and the cables were wrapped round both DEC and RA axis.

focaldepth, 2018-09-24:
kstars was actually kstars-bleeding

focaldepth, 2018-09-24:
I live at 54.5 degrees so IC1396 was close to the zenith at flip time.

knro, 2018-09-27:
Hmmm. shouldn't cable never wrap regardless of how the mount moved during the flip?

TallFurryMan, 2018-10-01:
If that could be related, I observe that sometimes the alignment module generates moves of ~360 degrees, instead of small positive or negative values. This reduces to the [-180,+180] interval with my own HEQ5-Pro on indi_eqmod_telescope, but could there be some drivers which interpret the raw value and do rotate as requested? This should be visible in the log in any case.

focaldepth, 2018-10-04:
Interesting TallFurryMan.
I intend to try and recreate the problem. I will grab some logs when I do so.
Do I need to turn on any extra logs?

I should get a chance next week.
We are off to star camp and will be out of range of the internet most of the time though.

focaldepth, 2018-12-03:
Apologies for the long delay. I have an explanaation and a work round for the problem.
I think that te issue was that the EQ "encoders" were out of sync.
I do not under stand why as they worked fine on Windows and EQ mod.
Manualy moving the mount and applying current did not fix the issue.
Manually moving, setting defaults and writing that did solve the issue.
So my theory is that park, goto and solve can work with the "encoders" out of sync but meridian flip can't.
Powering off the mount resets the encoders in the mount but not the saved park values.

I have repeated the issue on my EQ6 mount on a different fresh install of Kstars etc.

So I now have my portable rig and the Observatory both on Linux-Indi now.

Sorry but I have not had time to corrupt the encoders and get some logs.

focaldepth, 2018-12-03:
Apologies for the long delay. I have an explanation and a work round for the problem.
I think that the issue was that the HEQ5 "encoders" were out of sync.
I did not under stand why as they worked fine on Windows and EQMOD.
Manually moving the mount and applying current did not fix the issue.
Manually moving, setting defaults and writing that did solve the issue.
So my theory is that park, goto and solve can work with the "encoders" out of sync but meridian flip can't.
Powering off the mount resets the encoders in the mount but not the saved park values.

I have repeated the issue on my EQ6 mount on a different fresh install of Kstars etc.

So I now have my portable rig and the Observatory both on Linux-Indi. Great.

Sorry but I have not had time to corrupt the encoders and get some logs.
Spell checked!

I do not understand why a fresh install and a power on would not be in sync straight away.

So to recap, the solution is to manually park, default and write the encoders in site management tab.

TallFurryMan, 2018-12-05:
I completely, totally and fully agree with your last observation. I think that the encoder "issue" is a very important element of the understanding of how the mount works against astronomical calculus. The funny thing is that the longer one is using the mount, the more probable the situation will occur.
@focaldepth congratulations for understanding the issue on your own, I struggled with it when it happened to me this year too. There is a sticky forum thread to create here.

geehalel, 2019-01-07:
I wonder if the original problem is not that when you ask a counterweight-up goto, the driver stays in that state, and the subsequent meridian flip (managed by the client) is done as a counterweight-up goto ?
Anyway alignment never generates moves, it only alters the coordinates. A client may generate a meridian flip (a goto) based on the pier side property, which itself is based on the true telescope coordinates (not the aligned ones). That's the theory as I understand it. 
Now I still don't understand what you call the encoder "issue": it's very easy to corrupt encoders,  they are only relative thus the driver has to believe what you say. 
Concerning the problem of running the driver during a long period, it uses its own timer thus there may be some divergence. I don't know how to use ntp without setting the system time.

TallFurryMan, 2019-07-07:
@geehalel I need your advice here. I reproduced the "wrapping cable" issue on my HEQ5-PRO, and was saved by the horizon limits feature (which aborted the slew, but kept tracking on). Here is the log when it happened:

```
[           org.kde.kstars.indi] - EQMod Mount :  "[INFO] Starting Goto RA=20.2142 DE=38.412 (current RA=20.3204 DE=36.8789) "
[           org.kde.kstars.indi] - EQMod Mount :  "[INFO] GOTO ALign Nearest: delta RA = 0.106742, delta DEC  = -1.540432 "
[           org.kde.kstars.indi] - EQMod Mount :  "[INFO] Aligned Eqmod Goto RA=20.3209 DE=36.8716 (target RA=20.2142 DE=38.412) "
[           org.kde.kstars.indi] - EQMod Mount :  "[INFO] Setting Eqmod Goto RA=20.3209 DE=36.8716 (target RA=20.2142 DE=38.412) "
[           org.kde.kstars.indi] - EQMod Mount :  "[INFO] Slewing mount: RA increment = -4512119, DE increment = 2663323 "
[           org.kde.kstars.indi] - EQMod Mount :  "[INFO] Slewing to RA: 20:12:51 - DEC: 38:24:43 "
```

It appears the RA increment, -4512119, is larger than half the number of gear steps on the worm, 4512000. It may well be the reason for the wrong direction the mount rotates, but I have no idea yet why this happens. Any opinion?

[image]

knro, 2019-09-16:
@geehalel Any suggestions on how to proceed?

geehalel, 2019-09-17:
Sorry I don't know. CWUP goto ? The increment in DEC is also large (2663323, ~90 deg ?). Was this near the meridian (the goto may need a flip, after alignment).
I stopped astronomy for the moment and don't have looked at the driver updates, so can't be of much help.

focaldepth, 2019-09-18:
Hi, since reporting this, I have had many succsessful Flips and several failed ones.
Setting the hour angle at least 0.5 hours past meridian helps a lot.

All my failures seem related to using Plate solving.  My feel is that the Flip does nothing then plate solving goes crazy.

The log captured above gives a strong clue that the maths is not correct somewhere.

I have failed to capture useful logs. (many reasons).

I will try and collect some evidence in the logs  next time I have a clear sky.

A also intend doing a desk check on the software. (It may take a while to understand the logic though)

All time/weather/work/home permitting.

knro, 2020-02-24:
Alright, will close this for now due to inactivity. Please submit a new issue if you have any logs.

closed 2020-02-24 by knro