indilib/indi #717: indi_eqmod_telescope slewing on the wrong side of pier on second night

TallFurryMan, 2018-09-29:
When indi_eqmod_telescope has been running for 24h+, I observe an incorrect slew procedure for objects west and close to the meridian.

How to reproduce the issue:
- Start indiserver with indi_eqmod_telescope on night N.
- Use KStars to slew to an object west of the meridian, close to the meridian. In my log, this is M27 on 20180929 at my longitude. M27 is under 5 degrees of the meridian.
- Observe as indi_eqmod_telescope slews oriented on the east side of the pier, telescope pointing west.
- At end of observation night N, request indi_eqmod_telescope to park the mount, but do not stop indiserver nor indi_eqmod_telescope. In my test, I did not stop KStars neither. No computer did go to sleep.

[20180928_M27_EqModSlew_EastPointingWest_OK.log](https://github.com/indilib/indi/files/2430978/20180928_M27_EqModSlew_EastPointingWest_OK.log)

- On night N+1, use KStars to slew to the same object, which is now a little further west of the meridian. In my log, this is M27 again, approximately at 2 degrees west of the meridian.
- Observe as indi_eqmod_telescope slews oriented on the west side of the pier, telescope pointing east.

[20180929_M27_EqModSlew_WestPointingEast_KO.log](https://github.com/indilib/indi/files/2430977/20180929_M27_EqModSlew_WestPointingEast_KO.log)

For objects that are west of the meridian, under a few degrees, it is impossible to have indi_eqmod_telescope slew on the east side of the pier on night N+1. On night N+2, the limit is even further on the west.

I suspect the western limit under which objects are slewed to incorrectly corresponds to the position of the meridian on day N, shifted by the offset the sky sphere moves day after day. I didn't do the math though.

This issue can be "easily" worked around by stopping and restarting indi_eqmod_telescope. However, it points at a problem with the day-to-day offset, which accumulates over time and may cause meridian flips at unexpected times, and more seriously unwanted positioning when no meridian flip is planned. The HA of the meridian already shifts over a single night.

This is a serious preamble to tube hitting the tripod if the end-user is not aware of the position of the telescope when starting an observation.

TallFurryMan, 2018-10-20:
My problem is weirder than I thought. Tonight, I have an offset of roughly 4 hours in RA between the mount position and the position solved by Ekos.
I checked the mount itself with the synscan pad, it did gotos to main stars  roughly in the right direction. 
I checked the geographic position was identical in KStars and indi-eqmod. This might have explained the offset, but it would have had to be a large longitude difference.
I checked the UTC time was identical in KStars and indi-eqmod. I saw that there was a git change giving the clock to indi-eqmod only once when connecting. This might have explained a progressive shift if the remote system was drifting, but here the offset is large directly after connecting.
The issue currently causes all targets in the RA interval [meridian, meridian +4h] to be slewed to on the west side of the pier. And 4 hours is too large for my setup not to hit the tripod.

TallFurryMan, 2018-10-20:
I must note that this problem is recent. My last rebuild of indi-eqmod is beginning of September (ceda0c). I could image quite a few targets, but started noticing problems around the meridian from that point in time.

TallFurryMan, 2018-10-20:
The physical position of the mount matches the position of the solution in KStars. It means the position of the mount as represented by KStars is wrong by default. But I fail to see where the offset comes from. I'd say bluntly that the position represented by KStars use a UTC offset of -2h instead of +2h if it made any sense. But it doesn't to me.

knro, 2018-10-21:
Do logs shed any light on this? I have an EQ8, is there a way to replicate this reliably?

I guess one option is to git-bisect the problem until the offending code is found?

TallFurryMan, 2018-10-21:
Right now I'm trying to understand the situation. I did reset the firmware to factory settings. Eqmod driver is connected, geographic location is correct, LST is 20:04, telescope RA is 17:50, and KStars local meridian RA is 20:00. I don't event understand this.

TallFurryMan, 2018-10-21:
```
[SCOPE] Compute local time: lst=20.67858285 (20:40:42.90) - julian date=2458413.28255548
(...)
[ALIGNMENT] Status: Mnt. Algnt. NORTH Date 2458413.282555 encoders RA=6976821 DE=10647528 Telescope RA 18.433336 DEC 89.883511
[ALIGNMENT]  Direction RA(deg.)  33.678710 DEC 89.883511 TDV(x 0.001692 y -0.001127 z 0.999998)
```
The strange RA seems to result from the RA init step value. From the code, I understand that init value is taken at the connection stage. Disconnecting and reconnecting the driver shows me:
```
[DEBUG] GetRAEncoderZero() = 8388608
[DEBUG] GetRAEncoderTotal() = 9024000
[DEBUG] GetRAEncoderHome() = 8388608
[DEBUG] GetDEEncoderZero() = 8388608
[DEBUG] GetDEEncoderTotal() = 9024000
[DEBUG] GetDEEncoderHome() = 10644608
```
Now what happened to `RAEncoderHome`? This should be `6976821`!

TallFurryMan, 2018-10-21:
Except Home position and Park position are two different concepts. In my situation, I have a Home position and a Parking position that are different, but are pointing at the same North Celestial Pole.
How did I end up with this result? Several sessions with the scope hitting the tripod. Now RAEncoderHome and parking RA are really far away from each other.
My Parking counter points at the NCP, and my Home counter is assumed to point to the NCP by the driver: because those counter values are actually very different (6976821 vs. 8388608), the mount is still choosing the western side of the pier to slew up to 3-4 hours after the target passed the meridian.
Because this situation leads to an immediate hit on the tripod, the situation even worsens by itself by enlarging the offset even more.

TallFurryMan, 2018-10-21:
After all this investigation, the solution to recover this situation is very simple:
- Point the telescope to the NCP, either manually or by requesting parking (assumed the park location is NCP).
- Disconnect the driver.
- Modify the ParkData.xml file on the remote server, putting `8388608` as RA and `10644608` as DEC.
- Reconnect the driver.
- Check a slew east of the meridian is now solved close to its expected solution.
- Check a slew west of the meridian, outside of the problematic interval, is now solved close to its expected solution.
- Check a slew west of the meridian, close to the meridian, does result in the scope moving on the eastern side of the pier, and is now solved close to its expected solution.

closed 2018-10-21 by TallFurryMan

TallFurryMan, 2018-10-22:
Yes I think it does, you're right. One needs to press "Default", "Write File", then disconnect and reconnect the driver for the Home values to be used properly as zero step values.

TallFurryMan, 2018-12-05:
#713 is a similar issue.