indilib/indi #728: Adding PEC control to Celestron Nexstar

basilr, 2018-10-11:
It would be very helpful to have the ability to turn PEC on/off with INDI (not the training, uploading/downloading parts). The PEC status and control is available in newer version of the Nexstar hand controller.

I would be very happy to help in any way (I am not a programmer, but am comfortable with Linux, compiling, installing, and troubleshooting.)

I would be using this in Raspbian.

Thank you!

broadcastyourseb, 2018-10-15:
The PEC status and control is also known for older version of Nextar (In my case 4.21). I hacked my hand controller to know how it works. Same for me, i'm not a programmer, just a hobbyiste hacker.

Chris-Rowland, 2020-03-05:
I've done this. Can it be closed?

closed 2020-03-05 by knro