indilib/indi #738: Ekos appears to not be releasing ram after previews.

Blueshawk, 2018-10-23:
I've not been sure which way to go with this issue and waited for updates. The new distro (18.04) didn't clear it so I figured I better report it.
  I've been trying to narrow down a problem I've been finding that will eventually crash the system. When I use the focus module in ekos kstars ram usage goes up by each photo until it eventually reaches debilitating levels of ram used and operational lag results, then crashes to desktop. It also appears to be keeping previews in the capture module but I'm not positive of that one. I also haven't seen anything in logs leading to a problem, and can report that my previous file manager issues are not a part of it.

knro, 2018-10-23:
This is an Ekos issue which should be files under bugs.kde.org

Only INDI issues are reported here.

closed 2018-10-23 by knro

Blueshawk, 2018-10-23:
oh sorry. I wasn't sure which place to go with this and was under the misconception that indi and ekos were combined. I'll go check their issues then.