indilib/indi #743: indi_vantage driver don't update last 4 properties

mmastria, 2018-10-27:
This was discussed in forum:
https://www.indilib.org/forum/general/3407-vantage-pro-2-driver.html

I'm starting using this driver and verified that there's some bugs.
I corrected them and it's working on my observatory.

There's something else that I don't agree, but I'll make a new ticket as improvement ater bug corrections. It's about the use of internal temperature/humidity. There's no reason to use internal information, better using external temperature/humidity, as the telescope will be exposed to external conditions, and normally the receiver console (where the "internal" information is read) isn't in the same room, so this internal information have no real importance. But as I said, I'll put this in discussion in another ticket to not mix bug corrections with feature behaviour.

I'll create a branch to correct the bugs and a pull request on this.

DIff Code:

< OLD
> NEW

129c129
<     addParameter("WEAHTER_WIND_DIRECTION", "Wind Direction", 0, 360, 15);
---
>     addParameter("WEATHER_WIND_DIRECTION", "Wind Direction", 0, 360, 15);
268a269,275
>     // Inside Humidity
>     uint8_t humidityValue = loopData[11];
>
>     setParameterValue("WEATHER_HUMIDITY", humidityValue );
>
>     LOGF_DEBUG("Raw Inside Humidity (%d) [%#X4]", humidityValue, loopData[11]);
>
279c286
<     LOGF_DEBUG("Raw Wind Speed (%d) [%#X4]", windValue, loopData[14]);
---
>     LOGF_DEBUG("Raw Wind Speed (%d) [%#4X]", windValue, loopData[14]);

mmastria, 2018-10-27:
pull request:
https://github.com/indilib/indi/pull/745

knro, 2018-10-28:
Thanks! merged.

closed 2018-10-28 by knro