indilib/indi #748: AstroPhysics Experimental tracking indicator initial synchronization

wotalota, 2018-10-30:
On my system when Ekos connects to the AstroPhysicExperimental mount, the telescope panel comes up unparked and with the tracking button indicating off (the off button is highlighted red). However the mount is tracking. 
Clicking the Tracking off button puts up a 'are you sure' message box which when confirmed does not turn tracking off. 
Selecting tracking on, then tracking off does stop tracking.

knro, 2018-10-31:
@mikefulb Any ideas? Does the driver set SCOPE_TRACKING on startup if the mount was actually tracking initially?

wotalota, 2018-11-01:
The key to reproduce appears to be dropping power on the mount.
1. disconnect & shutdown ekos  
2. turn off and reapply power to the mount
3. restart ekos & connect
4. mount (in my case) is unparked. no indication that it is moving tracking off & alt az values at park 
   values.
Without the initial application of power starting/stopping park/unpark etc does not appear to reproduce.

Went through the above today to check a camera setting and forgot about it, remembered after 3 hours
and found it as in the screenshot.
[image]

closed 2018-11-01 by wotalota