indilib/indi #759: GPSD longtitude Error

yumoraby, 2018-11-09:
in xgps and gpsmon the lontitdue is correct, however in EKOS the longitude pulled from GPS module does not match. All other settings match information from GPSD, with exception of the longitude which is way off. 

Using the latest stable build 1.7.4 and kstars-bleeding. 

I have this issue replicated on an Intel, Raspberry Pi and a Libre Computer Le Potato. Using a U-Blox USB Gps module.

jochym, 2018-11-09:
Is the data in indi panel correct? Also try to run with C locale:
```
LANG=C indiserver ....
LANG=C kstars
```

yumoraby, 2018-11-09:
Yes the information is incorrect in Indi panel. The service is auto started using the indiwebmanager.service 

Do I run C locale from Bash?

If so would it be Lang=C indiserver -v

jochym, 2018-11-09:
So this is a problem with the driver. Yes, you run it from bash. 
The LANG variable is all caps. Like this:

LANG=C indiserver ......

yumoraby, 2018-11-09:
So enabled logging and can see that in the log file the location from device is obtained correctly. However when in indi control panel its incorrect.

jochym, 2018-11-09:
Did you try with LANG=C? Could you share the output (wrong/correct coordinates)?
Maybe by private message?

yumoraby, 2018-11-09:
>           Did you try with LANG=C? Could you share the output (wrong/correct coordinates)?
> Maybe by private message?

I tried I did not see any real difference whe n using LANG=C when running the server. I can share the log file output via DM.

jochym, 2018-11-09:
Did you run kstars with LANG=C as well? I have had some problems with this in the past.
Send the logs to jochym on gmail

yumoraby, 2018-11-13:
Logs sent. Did you get them?

jochym, 2019-01-17:
Sorry. I just missed them. They were sitting in my mailbox. I'll check.

jochym, 2019-01-17:
What should be correct longitude? 

And could you please run a minimal setup:
- gpsd linux driver
- indi-gpsd driver
- simple indi client like DCD, IPARCOS or RemoteObservatory - to eliminate possible kstars/ekos interactions?

If the last point is impossible run just this one driver and observe the behaviour with as much logging in the driver as possible and get those to me as well.

jochym, 2019-01-18:
Closing by @yumoraby request, since the problem seems to be due to different Longitude convention in programs.

closed 2019-01-18 by jochym