indilib/indi #763: syncan driver incorrect coordinates dec

didierd089, 2018-11-18:


didierd089, 2018-11-18:
the negative dec coordinates are taken for positive
ex synscan pro ra 0h11m53s dec -4d29m20s
indi api dec + 4d29m30s

knro, 2018-11-18:
Can you edit the code to add a couple of checks? in the synscandriver.cpp, at line 583 please add this
LOGF_INFO("n1: %ld n2: %ld ra: %g de: %g", n1, n2, ra, dec);

Then compile and reinstall, then reports what you get in INDI control panel when you run the driver.

didierd089, 2018-11-18:
[INFO] n1: 130130944 n2: 4080353024 ra: 0.727163 de: 342.011
ra 00h44m32 dec -17;53:07

knro, 2018-11-19:
Can you please git pull and test again?

didierd089, 2018-11-19:
no it's the same negative decs are positive in indi gui and in the planetarium

knro, 2018-11-19:
Can I get remote access to your system via TeamViewer or SSH so I can debug this remotely? If yes, please send me details to my email (mutlaqja AT ikarustech DOT com)

knro, 2018-11-19:
git pull it should be fixed now.

closed 2018-11-19 by knro