indilib/indi #765: QHY5-M- (SSAG) crashes when aborting exposure

downrange75, 2018-11-20:
Using the QHY5-M- as a guide camera crashes with the following on any attempt to cancel exposure:

`[   96.123571] usb 1-1: new high-speed USB device number 6 using xhci_hcd
[   96.249902] usb 1-1: New USB device found, idVendor=1856, idProduct=0011, bcdDevice= 0.00
[   96.249912] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[   96.304221] usb 1-1: USB disconnect, device number 6
[   98.044484] usb 1-1: new high-speed USB device number 7 using xhci_hcd
[   98.171024] usb 1-1: config 1 interface 0 altsetting 0 bulk endpoint 0x1 has invalid maxpacket 64
[   98.171317] usb 1-1: New USB device found, idVendor=16c0, idProduct=296d, bcdDevice= 0.00
[   98.171323] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   98.171328] usb 1-1: Product: QHY5-CMOS
[   98.171333] usb 1-1: Manufacturer: QHY_CAMS
[  152.110271] show_signal_msg: 18 callbacks suppressed
[  152.110276] indi_qhy_ccd[2856]: segfault at 2d0 ip 00007f5c5a781950 sp 00007fffc9cdf620 error 4 in libpthread-2.28.so[7f5c5a77e000+f000]
[  152.110294] Code: 48 89 7c 24 f8 64 48 8b 04 25 10 00 00 00 f0 48 0f b1 54 24 f8 c3 0f 1f 40 00 f3 0f 1e fa 41 56 41 55 41 54 55 53 48 83 ec 40 <44> 8b 87 d0 02 00 00 64 48 8b 04 25 28 00 00 00 48 89 44 24 38 31`

I modified the code to try the CancelQHYCCDExposing function rather than the CancelQHYCCDExposingAndReadout with no change.

Wrote the following patch and I was unable to reproduce the crash. Tested by running short exposures in a loop as well as single long exposures and then calling "stop" in Ekos. 

`--- qhy_ccd.cpp.orig	2018-11-06 17:47:13.480057481 -0500
+++ qhy_ccd.cpp	2018-11-06 17:53:52.641684587 -0500
@@ -929,17 +929,22 @@
     }
     pthread_mutex_unlock(&condMutex);
 
-    int rc = CancelQHYCCDExposingAndReadout(camhandle);
-    if (rc == QHYCCD_SUCCESS)
+    if (std::string(camid) != "QHY5-M-")
     {
-        InExposure = false;
-        LOG_INFO("Exposure aborted.");
-        return true;
+        int rc = CancelQHYCCDExposingAndReadout(camhandle);
+        if (rc == QHYCCD_SUCCESS)
+        {
+            InExposure = false;
+            LOG_INFO("Exposure aborted.");
+            return true;
+        }
+        else
+            LOGF_ERROR("Abort exposure failed (%d)", rc);
+ 
+        return false;
     }
     else
-        LOGF_ERROR("Abort exposure failed (%d)", rc);
-
-    return false;
+        return true;
 }
 
 bool QHYCCD::UpdateCCDFrame(int x, int y, int w, int h)`

knro, 2018-11-20:
Thank you. Can you please submit it as a Pull Request?

downrange75, 2018-11-22:
Commit https://github.com/indilib/indi/commit/ab48bdfbc3be1e404ddfc269cddd3e6169308fa7#diff-9dcfdfd865b356dfa00d842fb5f16b1b verified. Closing.

closed 2018-11-22 by downrange75