indilib/indi #773: IMAGE INFO

macicco, 2018-11-30:


macicco, 2018-11-30:
While setting up DSLR NIKON 40 DX, I am supposed to configure pixel pirch, image resolution on INDI - Image Info tab. But , any attempt to press set button after size inputs figures

macicco, 2018-11-30:
I tried manually set the size figures into file.xml, but even that did not work.

[image]

macicco, 2018-11-30:
It seems my issue is analog  to issue  #365

macicco, 2018-12-10:
> If you happen to use Ekos, then Ekos sets these values upon connection. You need to clear the DSLR values in order to re-enter them again.

Could you detail how clean values ? You mean, only "image info Tab" ?

knro, 2018-12-10:
Just press the "trash bin" icon next to the CCD name in capture module then restart.

closed 2018-12-10 by knro